mriscoc/Ender3V2S1 #585: Media Init Failed

bailed00, 2022-12-28:
After installing [Ender3V2-422-BLTUBL-T13-MPC-20221222.bin] everything is great except my microSD reader will not read any card I insert (media init failed) on both of my printers. I had the previous version working awesomely

jxer, 2022-12-28:
I have new issues with sd card as well. I have one printer going into endless reboot, until sd is removed , once it boots up its fine, I had 2 others crash during a sd card read as well.

bailed00, 2022-12-28:
I apologize for my not being more thorough in solving my own issue. It
appears that I had mixed in some exfat formatted cards. It definitely did
not like that so I reformatted to fat32 and the problem was solved. It
might be worth checking

jxer, 2022-12-28:
my sd cards need leave my machine. I also reformatted and moved cards. still is occuring

jxer, 2022-12-28:
but its not a media init fail, the printer doesn't load mesh, and goes into endless reboot, until the sd card is removed.

Amatsugu, 2023-01-03:
> I have new issues with sd card as well. I have one printer going into endless reboot, until sd is removed , once it boots up its fine, I had 2 others crash during a sd card read as well.

I have the exact same issue, Endless boot loop until sd card is removed. Sdcard is FAT32

GraemeDBlue, 2023-01-03:
I am getting the same , only happened after updating this firmware, I can see the following in the Octoprint terminal logs if this is helpful?

`Send: N0 M110 N0*125
Recv: start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Send: N0 M110 N0*125
Changing monitoring state from "Starting" to "Cancelling"
Recv:  Watchdog Reset
Recv: Marlin ProUI 2.1.3 MRiscoC
Recv: Build: 20221222 - 02:12
Recv: Author: Miguel A. Risco-Castillo (MRiscoC)
Recv: echo: Free Memory: 37015  PlannerBufferBytes: 1856
Recv: echo:V87 stored settings retrieved (760 bytes; crc 14128)
Recv: //action:notification Stored settings retrieved
Recv: Unified Bed Leveling System v1.01 inactive
Recv: Mesh loaded from slot 0
Recv: Mesh 0 loaded from storage.
Recv: //action:prompt_end
Recv: BL24CXX Check succeeded!`

codetak, 2023-01-03:
I have the same issue, I've been using the same SDCard without issues until the latest update. Endless boot loop until sd is removed. If "Media Update" is enabled, printer reboots when sd is inserted.

Workaround: Media Update=disabled, start printer without SD, insert SD

mriscoc, 2023-01-03:
> Recv:  Watchdog Reset

The kernel is waiting for an answer from the SD card but it is not responding, so a Watchdog Reset is issued.

codetak, 2023-01-03:
> > Recv:  Watchdog Reset
> 
> The kernel is waiting for an answer from the SD card but it is not responding, so a Watchdog Reset is issued.

It seems that randomly, after 4-5 reboots it manages to boot. Serial log bellow:

start
 Watchdog Reset
Marlin ProUI 2.1.3 MRiscoC
Build: 20221228 - 13:20
Author: Miguel A. Risco-Castillo (MRiscoC)
echo: Free Memory: 33911  PlannerBufferBytes: 1792
echo:V87 stored settings retrieved (754 bytes; crc 33932)
//action:notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
//action:prompt_end
BL24CXX Check succeeded!
start
 Watchdog Reset
Marlin ProUI 2.1.3 MRiscoC
Build: 20221228 - 13:20
Author: Miguel A. Risco-Castillo (MRiscoC)
echo: Free Memory: 33911  PlannerBufferBytes: 1792
echo:V87 stored settings retrieved (754 bytes; crc 33932)
//action:notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
//action:prompt_end
BL24CXX Check succeeded!
start
 Watchdog Reset
Marlin ProUI 2.1.3 MRiscoC
Build: 20221228 - 13:20
Author: Miguel A. Risco-Castillo (MRiscoC)
echo: Free Memory: 33911  PlannerBufferBytes: 1792
echo:V87 stored settings retrieved (754 bytes; crc 33932)
//action:notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
//action:prompt_end
BL24CXX Check succeeded!
start
 Watchdog Reset
Marlin ProUI 2.1.3 MRiscoC
Build: 20221228 - 13:20
Author: Miguel A. Risco-Castillo (MRiscoC)
echo: Free Memory: 33911  PlannerBufferBytes: 1792
echo:V87 stored settings retrieved (754 bytes; crc 33932)
//action:notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
//action:prompt_end
BL24CXX Check succeeded!
start
 Watchdog Reset
Marlin ProUI 2.1.3 MRiscoC
Build: 20221228 - 13:20
Author: Miguel A. Risco-Castillo (MRiscoC)
echo: Free Memory: 33911  PlannerBufferBytes: 1792
echo:V87 stored settings retrieved (754 bytes; crc 33932)
//action:notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
//action:prompt_end
BL24CXX Check succeeded!
start
 Watchdog Reset
Marlin ProUI 2.1.3 MRiscoC
Build: 20221228 - 13:20
Author: Miguel A. Risco-Castillo (MRiscoC)
echo: Free Memory: 33911  PlannerBufferBytes: 1792
echo:V87 stored settings retrieved (754 bytes; crc 33932)
//action:notification Stored settings retrieved
Unified Bed Leveling System v1.01 inactive
Mesh loaded from slot 0
Mesh 0 loaded from storage.
//action:prompt_end
BL24CXX Check succeeded!
echo:SD card ok
�echo: Serial 0 baud rate set to 250000
ok P15 B16

lookalek, 2023-01-17:
Hi guys,

I am also having issues with my SD card. Same as described above, sometimes after inserting the card, my printer falls into reboot loop until the card is removed. After a few trials, the printer finally reads the card correctly.

Also, sometimes inserting the card doesn't cause the reboot immediately, but it starts after entering the media folder on the main screen of the printer. Switching the printer on and off with a power switch also helps.

The card is formatted to FAT32. Recreating its partitions doesn't help either, same story.

BigApple13, 2023-01-17:
I was able to fix it by formating the sd card using windows. I used FAT32 and allocation size unit set to default.

lookalek, 2023-01-17:
> I was able to fix it by formating the sd card using windows. I used FAT32 and allocation size unit set to default.

Tried that many times, helps for a while only in my case.
Even used CMD's DiskPart to wipe the card completely and create a new FAT32 partition.

jxer, 2023-01-17:
So an oddity, my extruder fan was slowing failing, and the printer would reboot constantly with the sd card in. I changed the fan this weekend, and the issue has gone away. It obviously was some kind of emi rfi, issue, but ultimately, may be a timing issue on sd card loading. something definitely changed. changing the fan was an interesting fix.

SorenRR, 2023-01-18:
Try using Rufus to format your SD card ... here; [https://github.com/pbatard/rufus/releases](https://github.com/pbatard/rufus/releases)

I used Rufus to format a 64 GB PNY Elite Category 10 card as FAT32. My Windows 10 only offered exFAT !

jxer, 2023-01-18:
> Try using Rufus to format your SD card ... here; https://github.com/pbatard/rufus/releases
> 
> I used Rufus to format a 64 GB PNY Elite Category 10 card as FAT32. My Windows 10 only offered exFAT !

its not a card formatting issue, really need to stop saying this, my sd cards never change and I have 5 printers, only 1 was doing it.

SorenRR, 2023-01-18:
> its not a card formatting issue, really need to stop saying this, my sd cards never change and I have 5 printers, only 1 was doing it.

I'm using Octoprint now so I wouldn't know - except that Rufus can do miracles for a cheap SD Card.

RacerEXE, 2023-01-20:
For me reading the card fails unless I have "media update" enabled. When I enable the checkbox, it works fine.

mriscoc, 2023-01-25:
For general Q&A please post in https://github.com/mriscoc/Ender3V2S1/discussions

closed 2023-01-25 by mriscoc

lookalek, 2023-01-25:
> For general Q&A please post in https://github.com/mriscoc/Ender3V2S1/discussions

We are not having general discussion here but trying to describe the issue while eliminating external factors.
Issue is there, and is definitely not solved.