mriscoc/Ender3V2S1 #586: Thermal runaway protection system is bad

TheRealOne78, 2022-12-28:
So, the thermal runaway is supposeed to stop the printer whenever the thermistor reading is weird, to stop your 3D printer from burning your house.

It does work... but not very well. I'll explain:
So if you want to test the thermal runaway protection, one of the simplest way is to remove the sock and cool your nozzle block while warming up/hot. If the thermistor starts to send suspicious readings such like the nozzle sending a constant temperature while the nozzle still warms up, or the temperature instantly dropping.

So what doesn't work?
Here is a list of what works and what doesn't(if checked, it means it works)
* [ ] Starting to force cool the nozzle when starting to warm up
* [x] Starting to force cool the nozzle when 50% warmed
* [x] Starting to force cool the nozzle when fully warmed

Here are some videos showing my Ender 3 V2 when:
* [Cooling when starting to warm up](https://www.youtube.com/watch?v=qoPrx5_YsAI)
* [Cooling when mid warm](https://youtube.com/shorts/Im4pp7pB97Y)
* [Cooling when fully warmed up](https://youtube.com/shorts/URpz9-MutSE)

_PS: sorry I have to upload these videos on youtube, but I don't have any other alternative_

mriscoc, 2022-12-30:
I think that it is the normal behavior of Marlin firmware with the stock parameters of Creality configuration files.  As long as there is a positive change in the temperature when the nozzle is heated, no thermal runaway will be triggered.

TheRealOne78, 2022-12-30:
I can agree with you, but what if the temperature gradually rises but very slow? Wouldn't that be a sign of thermal runaway?

mriscoc, 2022-12-30:
Could also be an All-metal hot-end with a low power heater, a hot-end and part fan turned on and not good isolated nozzle, a filament with high specific heat. In any case an issue in the main Marlin repository could be better after trying the same configuration with the latest Marlin bugfix.

TheRealOne78, 2022-12-30:
Alright then, I guess I'll see more on the way.
Thank you for the info.

closed 2022-12-30 by TheRealOne78