mriscoc/Ender3V2S1 #587: MPC-AutoTune produces all "zero" parameters

gerdur, 2022-12-29:
Hello, I'm using here an 

- Ender3V2 
- MicroSwiss FullMetal-HotEnd with DirectDrive
- Briss Fang Duct with two NOCTUA fans 40x40x20mm fans for HotEnd- and Part cooling

System worked correctly  with all former professional firmware versions using HotEnd-PID. 

After installing Ender3V2-427-BLTUBL-MPC firmware version 20221222 the following issue happens:

1. When selecting "MPC-Autotune" with "HeaterPower=40.0" the HotEnd heats up to approx. 207°C
2. status changes to "Testing Heat Loss"
3. temperature curve flattens
4. after few seconds (before temperature descends?) the Finish Screen is shown "MCP Autotune done - Confirm"
5. all MPC parameter values except HeaterPower (Block capacity, Sensor respons. etc.) are showing "zero" values (0.000)
6. for me it looks like, the MPC Autotune process is aborted before successful finish.

**Expected behavior**
MPC Autotune process is running completely and calculating valid MPC parameter values.

**Version (please complete the following information):**
Ender3V2-427-BLTUBL-MPC-20221222.bin

**Additional context**
This issue was also reported in the Facebook "Creality Ender-3 V2 Professional Firmware Group"

mriscoc, 2022-12-30:
Maybe connecting the printer to a host as Pronterface or Octoprint terminal will give more information about the issue. Check also with https://marlinfw.org/docs/gcode/M306.html

Load default settings and try again, tested the latest release in an Ender 3v2 4.2.2 stock printer and MPC autotune works as expected.

gerdur, 2022-12-30:
Update:
OCTOPRINT shows the following during MPC-Autotune:

Recv: Cooling to ambient_
Recv: Heating to over 200C
Recv: Measuring ambient heatloss at 204.51
Recv: Temperature error
Recv: MPC Autotune finished! Put the constants below into Configuration.h
Recv: MPC_BLOCK_HEAT_CAPACITY 0.00
Recv: MPC_SENSOR_RESPONSIVENESS 0.0000
Recv: MPC_AMBIENT_XFER_COEFF 0.0000
Recv: MPC_AMBIENT_XFER_COEFF_FAN255 0.0000_

It looks like MPC-Autotune was aborted due to a "Temperatue error".

I'm quite sure that this is NOT a hardware issue:
After building identical professional firmware version 20221222 with BLTUBL and HotEnd-PID instead of MPC, everything works fine and all PID-Parameters are calculated correctly. 
Maybe there is a timeout triggered during "Heatloss testing", due to a larger "Block capacity" or something similar? (just an idea).

mriscoc, 2022-12-31:
Please try with the latest Marlin bugfix and if there is a problem report back the results here https://github.com/MarlinFirmware/Marlin/pull/23751

gerdur, 2023-01-01:
### Update2:
First of all, a happy and healthy new year to all of you!

Can someone please answer the following question for me:
After start of MPC-Autotune the HotEnd temperature rises to >= 200°C. When reaching this point, the status changes to "Testing heat loss" and a new target temperature is set (approx. 205°C in my case). Even if this target temperature is set, the HotEnd is still heatet more an more and does not stop at 205°C! 

**How is the behavior in a working system, escpecially when reaching state "Testing heat loss"? 
Is the HotEnd PWM switched off and the temperature curve flattens?**

In temperature.cpp I have seen that the MPC-Autotune function is aborted with an MPC_TEMP_ERROR, when the current_temp reaches "hotend.target + 15.0f" degrees. Just for test purposes, I changed this value to "hotend.target + 45.0f" and I'm observing now, that the hotend is heated up to 250°C, even if the target.temperature is still 205°C. Why does the heating not stop?

This is an OCTOPRINT terminal output of the test version BLTUBL_MPC with modified code and enabled debug messages:
Please notice, that now there are Non-Zero MPC parameters calculated, but the temperature error is still happening, so I guess these parameters are not reliable.

Recv: block temp -4294967295.42949672954294967295, celsius 237.33, blocktempdelta 6084969.50, delta_to_apply -1814217344.00, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
Recv: block temp -4294967295.42949672954294967295, celsius 238.27, blocktempdelta 12758598.00, delta_to_apply -3803941376.00, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 239.67, blocktempdelta 38736436.00, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 241.39, blocktempdelta 117607912.00, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 243.07, blocktempdelta 429674368.00, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 244.96, blocktempdelta 1304536576.00, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 246.50, blocktempdelta 3960710144.00, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 4294967295.42949672954294967295, pid_output 255.00, pwm 127
[...]
Recv: Temperature error
Recv: current_temp 248.01
Recv: t3 198.09
Recv: hotend.target 203
Recv: MPC Autotune finished! Put the constants below into Configuration.h
Recv: t1_time 33.00
Recv: sample_count 13
Recv: sample_distance 4
Recv: sample 0 : 100.63
Recv: sample 1 : 109.65
Recv: sample 2 : 118.59
Recv: sample 3 : 127.23
Recv: sample 4 : 135.57
Recv: sample 5 : 143.89
Recv: sample 6 : 151.86
Recv: sample 7 : 159.86
Recv: sample 8 : 167.85
Recv: sample 9 : 175.53
Recv: sample 10 : 183.25
Recv: sample 11 : 190.85
Recv: sample 12 : 198.09
Recv: t1 100.63 t2 151.86 t3 198.09
Recv: asymp_temp 1388.57
Recv: block_responsiveness 0.0017
Recv: MPC_BLOCK_HEAT_CAPACITY 17.27
Recv: MPC_SENSOR_RESPONSIVENESS -0.2988
Recv: MPC_AMBIENT_XFER_COEFF 0.0292
Recv: MPC_AMBIENT_XFER_COEFF_FAN255 0.0000
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 247.89, blocktempdelta 4294967295.42949672953066224750, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 0.00, pid_output 1.00, pwm 0
Recv: ok P15 B15
Recv: block temp -4294967295.42949672954294967295, celsius 249.12, blocktempdelta 4294967295.42949672954294967295, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 0.00, pid_output 1.00, pwm 0
[...]
Recv: block temp -4294967295.42949672954294967295, celsius 250.00, blocktempdelta 4294967295.42949672954294967295, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 0.00, pid_output 1.00, pwm 0
Recv: block temp -4294967295.42949672954294967295, celsius 250.08, blocktempdelta 4294967295.42949672954294967295, delta_to_apply -4294967295.42949672954294967295, ambient 13.68, power 0.00, pid_output 1.00, pwm 0

mriscoc, 2023-01-12:
The fan is turned on when the temperature is over 200 °C to around 210°C and MPC start to test how much heat is loss. That is the normal behavior of MPC.

gerdur, 2023-01-12:
Hello Miguel, many thanks for your reply. I suppose you are talking about the part cooling fan (the HotEnd fan is hardwired to 100% on)? 

In MARLIN, file temperature.cpp, at end of function MPC_autotune() (line 1062) I found the following code snippet:
  if (!WITHIN(current_temp, t3 - 15.0f, hotend.target + 15.0f)) {
    SERIAL_ECHOLNPGM(STR_MPC_TEMPERATURE_ERROR);
    TERN_(DWIN_LCD_PROUI, DWIN_MPCTuning(MPC_TEMP_ERROR));
    break;
  }

By placing debug messages at appropriate code positions, I detected that at this code position the MPC_autotune() function was aborted with MPC_TEMPERATURE_ERROR, when current_temp was reaching 220°C (hotend.target + 15.0f).

For debug purposes, I have increased the upper limit to "hotend.target + 45.0f" = 250°C. What I observed now was, that the HotEnd was heated up to 250°C, then the MPC_autotune() function was aborted with MPC_TEMP_ERROR, but now I had non-zero MPC_autotune parameter values! This was just a test, and so far I didn't check, if these nor-zero parameter values are valid and if MPC_autotune is working correct (remember: MPC_TEMP_ERROR is still active!). 

Intermediate solution so far is, that I am actually using professional firmware "Ender3V2-427-BLTUBL" with HotEnd-PID algorithm and this variant is working absolutely correct!

Giving the code snippet in function MPC_autotune() I suppose, that the HotEnd is heated up to 200°C, then the part cooling fan is activated, and then the HotEnd temperature is still rising, but it should NOT reach more than 220°C? For me, this sounds not plausible, as long as the HotEnd PWM is still active and HotEnd is still heated with up to 40Watts. Why does HotEnd heating NOT stop, after reaching the 205°C limit? 

I'm hoping the MARLIN MPC developer is reading this (I have seen your MARLIN issue report, linking to this thread. 
Many thanks so far and best greetings from Germany.

mriscoc, 2023-01-12:
I suggest you open an Issue in Marlin repository with your test.

mriscoc, 2023-01-25:
Since this firmware is a fork of Marlin, it is best to post any issues about the behavior of the Marlin kernel in the Marlin repository.

closed 2023-01-25 by mriscoc