mriscoc/Ender3V2S1 #589: homing fail

xcv826, 2023-01-01:
**Describe the bug**
any action that requires homing on the printer results in the printer saying "printer killed! homing failed! turn printer off."
**To Reproduce**
Steps to reproduce the behavior:
1. Go to any option that allows you to home.
2. initiate it.
3. it doesn't move or make any noises and get that immediate error.
4. See error

**Expected behavior**
for it to home.

**Version (please complete the following information):**
Ender3V2-427-BLTUBL-MPC-20221222.bin

JEBurroughs, 2023-01-01:
Advanced\Physical Settings\Park Head\ Here is were you set your home x y z
By default mine were like 220 220 10 or something like that, I always set this to 0,0,20 when I set my Machine's Physical setting after each new Versions. So if your Physical setting and or Mesh Inset is not correct then it's going to get crazy on you if the limits exceed a range.

xcv826, 2023-01-01:
> Advanced\Physical Settings\Park Head\ By default mine were like 220 220 10 or something like that, I always set this to 0,0,20 when I set my Machine's Physical setting after each new Versions. So if your Physical setting and or Mesh Inset is not correct then it's going to get crazy on you if the limits exceed a range. May mean nothing for your problem but at least worth a look and test.

didnt seem to help in the least bit, sadly. thank you for the help though.

any other ideas?

xcv826, 2023-01-01:
does my bltouch effect the firmware, in some way?

xcv826, 2023-01-02:
im a dumbass, i installed the wrong version lol. installed 4.2.7, instead of 4.2.2. everything is working now.

closed 2023-01-02 by mriscoc