mriscoc/Ender3V2S1 #594: Over 7x7 mesh probing issues when using Octoprint-BedLevelVisualizer

Matrixchung, 2023-01-02:
**Describe the bug**
When trigger mesh update from Octoprint, it can only resolve mesh points with 7x7 or lower, otherwise when probing at the 64th probing point, the printer will send a fatal error "Probing Failed" and kill the connection between host and printer.
The probing process could stuck at 64/81 Probing points at 9x9 size, and 64/64 Probing points at 8x8 size. 
However, smaller sizes of mesh points work well and the visualizer can show the graph correctly, and if you set the mesh points to over 7x7 but do the mesh building manually on printer without octoprint, it still come with correct result.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Advanced - Mesh Leveling` and set `Mesh Points` to over 7x7
2. Click `Update Mesh Now` in Octoprint-Bed Visualizer
3. Wait until the probing process comes to 64th point
4. See error, printer killed and broke the serial connection

**Expected behavior**
It's expected to behavior the same as 7x7 mesh points or lower.

**Version (please complete the following information):**
Ender3S1-F4-UBL-20221222

**Additional context**
My settings of Bed Level Visualizer:
```
M190 S60  ; (optional) wait for the bed to get up to temperature
M155 S30  ; set temperature reporting delay, use a value longer than the time it takes for your leveling command to complete.
G29 P1    ; Take a bed leveling mesh.
@BEDLEVELVISUALIZER  ; instruct plugin to start recording responses from printer.
M420 V    ; Get bed leveling mesh data.
M155 S3   ; set the temperature reporting delay back to a shorter time span.
M500      ; save the current setup to EEPROM
M140 S0   ; cooling down the bed
```

mriscoc, 2023-01-02:
Maybe: https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-crtouch

Matrixchung, 2023-01-02:
> Maybe: https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-crtouch

Update: If I clear all mesh and build new 7x7 mesh with Octoprint, it still work well. So it seems that the problem only occurs on larger mesh points, and has nothing to do with the `M420`  command.

feuerblitz2, 2023-01-04:
I can also confirm the problem.
Ender3s1
Ender3S1-F1-UBL-MPC-20221222.bin
9X9 Mesh
before the measuring point ber bl touch is pulled in then the print head crashes on the bed. In Oktoprint comes ne error message. directly at the Ducker machter further with the next point. but then all the following Messpunte shift that at the end one is missing.

vrooze, 2023-01-04:
Hey

i use 9x9 on my Ender 3 v2 Neo on 4.2.2.
I noticed this wasn't working also but after closer inspection, I noticed there is no longer an non-UBL version.
If, like myself, you had the non-UBL release last time you need to swap the  config in octoprint a bit:

Non-UBL version:
``` 
--- snip ---
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting
--- snip ---
``` 
for the UBL version:
```
--- snip ---
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
M420 S1 V ; enabled leveling and report the new mesh
--- snip ---
```

That at least has mine actually probing now!

Hope this helps!

V

MarkforU, 2023-01-04:
Always has this error and problem with Creality CR-Touch, it couldn't handle the high speed probing. Always retraction of the probe around the same area on my Ender 3 V2 with 4.2.2 board. Usually it would crash the bed around 64-84 when doing a 9x9. Turning off HS Probing and no more issues. I've since bought a Creality BL-TOUCH and will test it today to see if I have problem with HS on a new Machine coming in. I'm going to guess it will have a problem because the BL-TOUCH is Creality version...think if I have the same problem, I'll send it back and pick up a BL-TOUCH from the other original manufacturer of them.

mriscoc, 2023-01-04:
Not a firmware issue, it is a CRTouch issue and the solution is already written in the Wiki guide.
 https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-crtouch

mriscoc, 2023-01-04:
Duplicate of: https://github.com/mriscoc/Ender3V2S1/issues/485, https://github.com/mriscoc/Ender3V2S1/issues/203, https://github.com/mriscoc/Ender3V2S1/issues/32, https://github.com/mriscoc/Ender3V2S1/issues/85

closed 2023-01-04 by mriscoc

Matrixchung, 2023-01-05:
Disabling the HS mode does help. Sorry for making duplicate issue and thank all of you!
Additionally, Creality CR-Touch with the HS mode usually gets a high repetitive error, so if anyone is experiencing that problem, disabling high speed probing may also help.