mriscoc/Ender3V2S1 #595: Z Axis Falling

I-wish-2, 2023-01-02:
**Describe the bug**
This problem occurs on 3 of my printers flashed with the Ver 2.1.3 20221002 firmware.  All are Ender 3 Pros with the 4.2.7 M.B. Two have the Z axis timing belt upgrade ( YouTube Kevin Akasam ).  One has the original T8 leadscrew.
  
At the conclusion of the End G-Code after a successful print all the Steppers are disabled even if the End G-Code is not calling for them to be disabled.  It will also occur after the initial power up of the printer about 2 minutes after homing using the Prepare screen.

This will cause the printers with the timing belt Z axis gantry to free fall to print bed impacting the nozzle on the bed.  The leadscrew printer does not fall but will drift down over about 15 minutes.  My End G-Code is below:

; Ender 3 & 5 Custom END G-code 12/12/2022
; This code is based on the E3 V2 Klipper setup buts work on all E3 & 5 Printers

G91		;Relative positioning
G1 E-2 F2700	;Retract 2 mm
G1 E-2 Z0.2 F2400	;Retract and raise Z
G1 X5 Y5 F3000	;Wipe out
G1 Z10		;Raise Z more
G90		;Absolute positioning

G1 X10 Y225 Z75	;Park X and Z axis; Move Y to Present Print
M106 S0		;Turn-off fan
M104 S0		;Turn-off hotend
M140 S0		;Turn-off bed

M84 X Y E 	;Disable all steppers but Z to lock the gantry at 75mm

; End of custom END G-Code

**To Reproduce**
Steps to reproduce the behavior:
1. Power up the printer
2. Select " Prepare " then home all axis.  Then raise the Z axis 100mm
3. Wait about 2 minutes and all axis will be disabled.
4. Test by hand moving the axis.

1. Start a normal print.
2. At the completion of the End G-Code all axis will be disabled.

**Expected behavior**
Do not disable the steppers at any point unless the G-Code call for it.

**Screenshots**
None.

**Version (please complete the following information):**
Ver 2.1.3 20221002  427- BLTUBL

**Additional context**
Thank you for reviewing and improving this great firmware.

mriscoc, 2023-01-02:
For security, motors and heaters are disabled, but it is possible to compile a customized version by changing this function:
https://github.com/mriscoc/Ender3V2S1/blob/72b609366965ba83826b6124878193de20c3d664/Marlin/src/lcd/e3v2/proui/dwin.cpp#L1621-L1630

I-wish-2, 2023-01-02:
Thank you Miguel for clarifying the reason for disabling the steppers and heaters as well the way to customize the firmware.  I just wish I had a clue as to how to edit and compile it.  It is way over my head at this point in time.

Best Regards
Doug

I-wish-2, 2023-01-02:
Hi Miguel,  Using Visual Basic Code I was able to navigate to dwin.cpp then to lines 1621 thru 1630 but sorry to say I have no idea as to how to edit it or then compile it.  But I'm willing to try.
Doug

I-wish-2, 2023-01-02:
Would commenting out the line " planner.finish_and_disable(); " be the edit I need?
Thank You again
Doug

I-wish-2, 2023-01-03:
Hi Miguel,  I commented out line 1627 in the dwin.cpp file and recompiled successfully.  After flashing the printer would not home.  After flash return to your current version the printer functioned ok.  I'm going to move this information to the Discussions area where I think it needs to be.
Thank you for advancing my understanding and for the "LIVE MOVE " feature in the current version.
Doug

closed 2023-01-03 by I-wish-2