mriscoc/Ender3V2S1 #606: Bed tramming with probe unreachable mesh points

Glandvador, 2023-01-03:
**Describe the bug**
Defining a mesh with unreachable points by the probe affects the bed tramming functionality, aka some corners are not probed.

**To Reproduce**
Steps to reproduce the behavior:
1. Define a mesh inset as 10,10,210,210
2. Don't maximize area.
3. Do a mesh build, some point are not probed because out of probe reach (normal behaviour). Manually completing the missing points is not needed.
4. Try the bed tramming process and only some points (ones reachable by the) are tested.

**Expected behavior**
Bed tramming should reach all the corners. 

**Version (please complete the following information):**
 Ender3S1-F4-UBL-20221222

**Additional context**
Doing a mesh tilt works by using the maximum coordinates reachable by the probe. I think that bed tramming should do the same. I have modified the code (as a proof of concept):

diff --git a/Marlin/src/lcd/e3v2/proui/dwin.cpp b/Marlin/src/lcd/e3v2/proui/dwin.cpp
index 785b0ab..9f55d5d 100644
--- a/Marlin/src/lcd/e3v2/proui/dwin.cpp
+++ b/Marlin/src/lcd/e3v2/proui/dwin.cpp
@@ -2426,8 +2426,18 @@ TERN(HAS_BED_PROBE, float, void) Tram(uint8_t point) {
       queue.inject(cmd);
     }
     else {
-      LIMIT(xpos, MESH_MIN_X, MESH_MAX_X);
-      LIMIT(ypos, MESH_MIN_Y, MESH_MAX_Y);
+      float mesh_min_x = MESH_MIN_X;
+      float mesh_max_x = MESH_MAX_X;
+      float mesh_min_y = MESH_MIN_Y;
+      float mesh_max_y = MESH_MAX_Y;
+#if ProUIex
+      mesh_min_x = _MAX(mesh_min_x, probe.offset.x);
+      mesh_max_x = _MIN(mesh_max_x, X_MAX_POS + probe.offset.x);
+      mesh_min_y = _MAX(mesh_min_y, probe.offset.y);
+      mesh_max_y = _MIN(mesh_max_y, Y_MAX_POS + probe.offset.y);
+#endif
+      LIMIT(xpos, mesh_min_x, mesh_max_x);
+      LIMIT(ypos, mesh_min_y, mesh_max_y);
       probe.stow();
       gcode.process_subcommands_now(F("G28O"));
       inLev = true;

In a perfect world, I would have copied the code from maximize area method, but unfortunately it exceeds the stmf4 .data size. Maybe there is a better way to do that, but I don't know the Marlin code.
Normally, this modification should have no impact on people using a maximized/centered mesh area.

BTW, I think that I spot a copy/paste error, that has no real consequence:

diff --git a/Marlin/src/lcd/e3v2/proui/dwin.cpp b/Marlin/src/lcd/e3v2/proui/dwin.cpp
@@ -1724,9 +1724,9 @@ void DWIN_SetDataDefaults() {
     #if HAS_MESH
       PRO_data.grid_max_points = DEF_GRID_MAX_POINTS;
       PRO_data.mesh_min_x = DEF_MESH_MIN_X;
-      PRO_data.mesh_max_x = DEF_MESH_MAX_Y;
+      PRO_data.mesh_max_x = DEF_MESH_MAX_X;
       PRO_data.mesh_min_y = DEF_MESH_MIN_Y;
-      PRO_data.mesh_max_y = DEF_MESH_MAX_X;
+      PRO_data.mesh_max_y = DEF_MESH_MAX_Y;
     #endif
     #if HAS_BED_PROBE
       PRO_data.zprobefeedslow = DEF_Z_PROBE_FEEDRATE_SLOW;

mriscoc, 2023-01-03:
Mesh insets define the work/leveling area, so the tramming points are configurable by modifying mesh insets. You can maximize the work/area or center that area in the mesh inset menu, but for this to work as expected, the XY MIN/MAX position must be defined using this guide: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides
The general guide is, tramming points are on the corners of the work/leveling area.

mriscoc, 2023-01-03:
> BTW, I think that I spot a copy/paste error, that has no real consequence:
>
> diff --git a/Marlin/src/lcd/e3v2/proui/dwin.cpp b/Marlin/src/lcd/e3v2/proui/dwin.cpp `@@ -1724,9 +1724,9 @@ void [...]

Yes, thanks, that error is currently fixed but was not published in the NY2023 release.

closed 2023-01-03 by mriscoc