mriscoc/Ender3V2S1 #611: CR Touch doesn’t work and the nozzle crashed into the bed always

mobigroup, 2023-01-04:
It looks the probe is not supported in the code because nozzle moving by z axis without any limit and probe sensor is even not activated. Probably, firmware waiting for a signal from non existing z limit switch which is not compatible with CR Touch and so it doesn’t exist in CR Touch configurations. See for details #235 Also, the ticket relates to #608 - is it by design the only way to use a probe by g-code via serial interface? Actually, I don’t understand the situation because probe support is declared for the project but it doesn’t work completely and the maintainer closed the related discussion #235. Obviously, original Marlin firmware has probes support and this feature is important.

djarmag, 2023-01-04:
Solder and install a 10K resistor 
https://www.youtube.com/watch?v=gGDSVacRgJo

I had same intermittent issues on both 4.2.2 and 4.2.7 and this fixed it permanently.

mobigroup, 2023-01-04:
@djarmag how are you going to fix the firmware issue by this way? The hardware is ok and it works on Creality firmware so your suggestion is the way to lose the probe warranty without any reason. Maybe the firmware compiled without the probe support.

mriscoc, 2023-01-04:
If your CRtouch is wired correctly then the problem could be the interrupt services for the GD being different than the STM. If that is the case the support for GD chips will be deprecated. But we need to wait if anyone else confirms that behavior.

djarmag, 2023-01-04:
@mobogroup How is it voiding warranty? It's preventing false triggers from happening and is soldered on board side, never physically altering the touch itself. Plus, the same non-trigger and false triggers were even happening with BLtouch. I've recommended this solution to 4 other people having similar issues and they all worked since then as well. 
@mriscoc if there is something you'd like me to try I would be willing to test it out on my printers

mriscoc, 2023-01-04:
> @mriscoc if there is something you'd like me to try I would be willing to test it out on my printers  

Thanks, seems boards with GD SoC are problematic, but I need to diferentiate intrinsic problems, from erros caused by bad wiring/installation

mobigroup, 2023-01-10:
@djarmag But soldering on board side voiding mainboard warranty of course. And you missed the main point - the sensor works on stock firmware while auto leveling is totally broken (I cannot print after auto leveling because nozzle crashes into bed) due to garbage Creality firmware for GD mainboard.

mobigroup, 2023-01-10:
@mriscoc The sensor is functional on the stock firmware. And yes, the chip frequency is different so STM config

closed 2023-01-10 by mobigroup

mobigroup, 2023-02-04:
@mriscoc I've replaced the mainboard to STM-based Creality 4.2.7 and the CR Touch sensor works on mriscoc firmware.

Some users report their GD32F303RET6 boards bought earlier work with CR Touch using Marlin firmware! But the processors look different to the installed on my board by default - so I suppose there are some different incompatible variations GD32F303RET6 processors actually.  

By the way, I've got a partitial refund for the mainboard replacement via aliexpress support due to the GD instead of STM mainboard installed - there is no Creality Filament sensor supporting firmware on  Creality downloads site and the firmware for the board named as "GD Ender 3 v2" on the official Creality site so it does not the same one which we expect for Ender 3 v2 printer. The seller is not able to provide firmware to connect my Creality filament sensor for Ender 3 v2 to the Creality Ender 3 v2 printer and that means the printer has unsupported mainboard. Also, CR Touch on the GD board works as Z-limit switch only and it's impossible to print after auto-leveling... and all the terminal commands to check the sensor are disabled. I disabled CR Touch for Creality firmware on GD board to be able to print using manual leveling. The GD board is useless although it works (sure, not well) without CR Touch and filament sensors.

kenny00111, 2023-02-06:
I have a GD board and it works with the Cr touch, filament sensor enabled

mobigroup, 2023-02-06:
@kenny00111  Could you share your processor and mainboard photo to compare?

kenny00111, 2023-02-06:
[image]
Hopefully it is visible. But I didn't read the title, I use the normal STM firmware

mobigroup, 2023-02-06:
@kenny00111 Yes, it is well visible. This is my one with the same T8 stepper drivers code which does not work with CR Touch using mriscoc and other marlin-based firmwares:

[image]
[image]

The mainboards look different while have the same version number. I don't see if your GD processor has the round notch near to "Giga" text on it?

kenny00111, 2023-02-06:
Yes it has one

mobigroup, 2023-02-08:
@kenny00111 Maybe did you try to install original Creality firmware for STM processor (without “GD” prefix) - does it work for you? For me it doesn’t work.

sammcj, 2023-02-12:
Here's my GD32F303 (It's all working fine for me).

[image]