mriscoc/Ender3V2S1 #614: Ender3S1 F4 ABL not compensating

needs-coffee, 2023-01-04:
I have noticed that the ABL when used with my ender 3S1 and the professional firmware does not appear to apply compensation during the prints to account for the bed levelling.
Before each [test print](https://www.printables.com/model/45752-3x3-bed-level-calibration) below i trammed the bed using the tramming wizard - each time there were all 0 except for the bottom right which was -0.01. 
The Gcode was produced using cura with a 0.2mm layer height. The `M420 S1 Z5` was present in the start G-code after `G28` and only removed for the no abl print. During the ABL prints the z-offset lines were blue and the icon blinking as per the Wiki (not blinking and white during the non-abl print).

It appears for each print the top left is consistently too high and bottom right too low. 
I have checked the height of the X gantry above the bed and frame and these are the same left and right. 

WITH 5x5 ABL
[image]

WITH 4x4 ABL
[image]

NO ABL
[image]

This issue is present: 
- on both `professsional-Ender3S1-F4-20221002.bin` and the current firmware `Ender3S1-F4-UBL-20221222.bin`
- Printing from octoprint and via SD
- Multiple probing 0 1 or 2
- ABL Mesh size 3x3 up to 8x8
- All have been tested with `HS Mode` enabled - I have yet to try with this option disabled. 
- Multiple different print files not just the calibration example above. 

I am at a loss as what to do to improve the first layer of my prints with the current firmware. From the prints above i don't see any difference in the print output when ABL is enabled or disabled. 

Any suggestions on what to try ?

Many thanks for your work on this firmware!

mriscoc, 2023-01-04:
Possible duplicate of https://github.com/mriscoc/Ender3V2S1/issues/195 and https://github.com/mriscoc/Ender3V2S1/issues/198

mriscoc, 2023-01-04:
Currently some S1 (F1 and F4) have issues with the CRTouch, sadly it seems to be a hardware problem. I suggest you use the automatic leveling as reference and then do a mesh editing and save to two slots (one for backup). That will give you the best results.

needs-coffee, 2023-01-04:
> 

> m. I suggest you use the automatic levelling as reference and then do a mesh editing and save to two slots (one for backup)

thanks for your reply

Interesting - this does seem similar to #198 
I will try manually editing the mesh later and see if this improves/changes things

do you mean this is a hardware issue with the printer or with the CR Touch probe? 

the probe does appear to consistently measure the same points on the mesh when the auto mesh is repeated, but the corrections 
from this mesh profile is not applied during the actual print

mriscoc, 2023-01-05:
After printing can you confirm in the mesh viewer if you have a valid mesh in memory?
Also verify that you see the z-offset icon change when leveling is active:

https://user-images.githubusercontent.com/2745567/210678467-308f3f9c-9c05-4ef8-8824-d4e48a072bc0.mp4

needs-coffee, 2023-01-05:
Mesh is saved in memory after prints and after reboots. 
The Z-offset blinks as per the video and wiki when printing with the ABL enabled. 
HS mode and changing multi probing doesn't change the produced mesh. 

I also tried creating the mesh with the z-offset set to zero and to the correct offset as per reports in #195 
This did not change the produced mesh beyond what would be expected from margin of error.

I created a new print with a manually edited mesh and as can be seen below the post edited mesh is quite different from the pre edited mesh and the print is not perfect, but much better.  With these prints it does seem the saved mesh is applied as corrections to the print job, but the issue is with the accuracy of the automatic mesh which is puzzling and frustrating. 

AUTO MESH
[image]

MANUAL MESH
[image]

AUTO MESH PRINT
[image]

MANUAL MESH PRINT
[image]

mriscoc, 2023-01-05:
In one of the referenced threads a user explains that some F4s have a little twist in the X profile that causes a little deviation because when the CRTouch takes the metering, the nozzle is in another XY position, I know that some users print an adapter for move the CRTouch to the right side fo the head to set the Y Probe offset to 0. I first thought the problem was about the F4 SoC, but I've been getting reports from people with F4 with no mesh issues.

mriscoc, 2023-01-05:
This could help:
https://www.printables.com/model/213992-perfect-1st-layers-ender-3-v2-ender-3s1-ender-3s1-

grrmisfit, 2023-01-08:
> > I can comfirm that "something" is very off with the CR touch. (Ender 3 S1, F4 controller) Semms to be the bigger the Mesh the bigger the offset, at least for me.
> 
> From the picture, your bed is not thermally stable, wait a bit and take a new mesh, the points close to the Z home position should be close to 0.

actually its the probe mounting. this mod https://www.printables.com/model/257460-ender-3-s1-zero-offset-cr-touch-mount and changing Y offset to 0 then redoing my z offset fixed all issues the OP has stated. I had exact same issues and only manual mesh would let me print, then i was pointed out this mod for the S1 ( there is a pro version by someone else) but now my mesh now matches when i verify each point on mesh edit, where before this mod it did not. 
Could it be something in marlin isnt using the Y offset right and thus when set to 0 it fixes it?

experiment322, 2023-01-08:
> > > I can comfirm that "something" is very off with the CR touch. (Ender 3 S1, F4 controller) Semms to be the bigger the Mesh the bigger the offset, at least for me.
> > 
> > 
> > From the picture, your bed is not thermally stable, wait a bit and take a new mesh, the points close to the Z home position should be close to 0.
> 
> actually its the probe mounting. this mod https://www.printables.com/model/257460-ender-3-s1-zero-offset-cr-touch-mount and changing Y offset to 0 then redoing my z offset fixed all issues the OP has stated. I had exact same issues and only manual mesh would let me print, then i was pointed out this mod for the S1 ( there is a pro version by someone else) but now my mesh now matches when i verify each point on mesh edit, where before this mod it did not. Could it be something in marlin isnt using the Y offset right and thus when set to 0 it fixes it?
> 
> edit: I have 2 S1s with F1 chips and both have this mod and had those issues, I have read that F4 chip might have other issues but it doesnt hurt to try.

This fixed the issue for me too. Not sure if the issue is with marlin, but what I suspected was that the Y axis was bent (the V guide - the place where the wheels roll - or the wheels were deformed in some places?) and thus had a different offset when the nozzle reached the Y offset probed by the CR Touch because of it. Never really confirmed it, but I was so relieved after so much headaches I didn't bother anymore because I had enough of it.

vfrt123, 2023-01-09:
> actually its the probe mounting. this mod https://www.printables.com/model/257460-ender-3-s1-zero-offset-cr-touch-mount and changing Y offset to 0 then redoing my z offset fixed all issues the OP [...]

grrmisfit, 2023-01-09:
> Many thanks! Finally a working solution to this problem. Ideal automatic first coat. I had to edit the mesh with paper. I think there is a problem with this firmware. (Ender3S1 F4)

I don't think so, unless it's a marlin bug itself, as I had this issue on another firmware not pro. Best I can offer anyone atm is edit mesh for each point till you can print this mod. But again make sure you edit probe y offset to 0 and set z offset to 0 then redo your z offset as this mod also cut my z offset in half

Rabjerg, 2023-01-09:
> From the picture, your bed is not thermally stable, wait a bit and take a new mesh, the points close to the Z home position should be close to 0.

I retested and with the bed manually leveled, then brought up to temperature over the whole surface, the bed was still level. 
While at temperature i did the ABL again, and it came to the conclusion that the corners was ca. 2mm lower than center. 
With a steel ruler i checked if the bed was warped, and din't see a 2mm bump anywhere. 

I'm currently printing the new backplate to see if that might help.

MarkforU, 2023-01-09:
I'm running latest Marlin Firmware version (New Year Update. I'm using 4.2.2 board on my E3V2 with CR-Touch. I'm doing 9x9 bed leveling, PEI Flex Steel. I have a excellent mesh being put down. In the end, I show offset between 0 and -.12 on my bed, tramming shows same number in the 4 corners that my tramming comes up with. I'm running UBL version of firmware. When I go to print something like the Chinese Dragon which uses pretty much the whole bed because it is 2ft 3in long, my first layers are perfect, just like it was with previous version, which was the Spring Release. I've Even gone to the point of swapping my CR to a BL-Touch and I come up with exactly same numbers. Nothing is crazy high or low as I've seen in other pictures, I'm also using the Cura Calibration shapes and using the bed leveling template and it's perfect, no high or low points that the nozzle would drag or miss. I don't use springs or silicone spacers because my bed is locked down between 2 sets of wheels in place of springs. So, I'm not seeing these dramatic changes between all 4 of my machines which 3 run CR-Touch and one running BL-Touch with high speed probing on that one. If I turn the on High Speed Probing on the CR-Touch, I will get inconsistent probing numbers and during the 81 points, usually around the 64 point range my probe will not deploy and the nozzle will come down and hit the bed and push off, then deploy probe on that spot and then probe it, when it finishes, if I don't get a error that halts the machine and asks to restart, my numbers will be off drastically. Not sure if this helps? When I installed the firmware and went through Menu's to setup each machine, I've noted on installing firmware that HS Probing is checked under that menu, I have to go into 3 of the machines and turn it off, on the BL-Touch I leave it on. Machines preform perfectly with all 4 being able to print the same Bed Level Calibration test offered through the Cura Shape Calibration plug-in. So, I don't see the issue, unless, like I said, if I disable the HS Probing on the CR-Touch. I went and looked at the mod file that was posted in the link above which looks like it lowers and offsets the nozzle a bit, but I'm not needing it. Saturday, I helped a friend set his machine up, almost same configuration, he's using springs, but his mesh looks like a punch bowl with green in center and red around the outside, he's also running the same 4.2.7 firmware with UBL. I disable his high speed probing, his mesh was off on the left front of the bed doing a 9x9 grid, but when we went back and tried a tramming wizard it also confirmed that the left side had shifted up, so we had to tighten up his springs tighter and run tramming wizard again and by the time we ran another mesh, his bed is still showing a punch bowl effect, but he was able to also use the Cura Calibration Shapes plug-in and we put down a smooth first layer. I didn't check his with steel ruler to see if it showed any high or low points, just each time we trammed the bed showed it was 0 in center and the outer 4 corners were a .2 off from center, the mesh shows the as the Sam mainly close to 0 in center and .20 on the outsides of the 9x9 mesh.

grrmisfit, 2023-01-09:
> > > I can comfirm that "something" is very off with the CR touch. (Ender 3 S1, F4 controller) Semms to be the bigger the Mesh the bigger the offset, at least for me.
> > 
> > 
> > From the picture, your bed is not thermally stable, wait a bit and take a new mesh, the points close to the Z home position should be close to 0.
> 
> I retested and with the bed manually leveled, then brought up to temperature over the whole surface, the bed was still level. While at temperature i did the ABL again, and it came to the conclusion that the corners was ca. 2mm lower than center. With a steel ruler i checked if the bed was warped, and din't see a 2mm bump anywhere.
> 
> I'm currently printing the new backplate to see if that might help.
> 
> Update: After printing, mounting and measuring the offset of the new mount i get a lot more plausible mesh. [image]

glad it helped. trying to get the word out, its not my design but man i wish someone would have told me about this before as it would have saved me a lot of headaches.

TheWebMachine, 2023-01-10:
> actually its the probe mounting. this mod https://www.printables.com/model/257460-ender-3-s1-zero-offset-cr-touch-mount and changing Y offset to 0 then redoing my z offset fixed all issues the OP [...]

OMG! This needs to be mentioned in the Wiki under caveats for the CRTouch probe! Printing this mod (on my trusty Prusa, mind you) to move the probe worked perfectly on my S1 V2 Plus (F4 chip) and now my test prints line up to the mesh without a single issue! I was even able to re-enable HS mode and keep the 960 Z Feed with only 0.0015 deviation on the M48 test. In short, moving the probe fixed all the issues I was having.

I have no idea why the probe mod fixes things, whether it be the damn wheels and rails of the Y-axis where you can never get the wheels to exactly the right tension or a bug in the firmware (doubtful unless the same bug is in Creality stock firmware/Marlin where I had the same problem) or something else entirely. All I know is, moving the probe fixed an issue that has been plaguing me with this printer for the entire 3 weeks I've had it! 

I will note that I had to set the probe offsets a little differently on my Plus model than what the Printables page said once I got it installed. I measured it and got a probe offset of -45.4x and -1.8y. Don't bank on using the listed values on the page and measure for yourself, if you can.

jongdesteven, 2023-01-10:
I was too struggling with the ABL on S1 with CRtouch.

I removed my print bed and disconnected the belt for the Y-axis. there where obvious rough points.
After loosening all 6 bolts for the profile, all of them where super tight, and torquing them with limited force i see my results are WAY better. 
The 0-offset probe mount eliminates the difference in measuring position vs print-position on the Y-axis.
Fixing the actual Y-axis is another route, maybe both will be ideal!

wibenjamin, 2023-01-20:
I think I can confirm this is **not** an issue with this firmware, but at least for my machine an issue with the hardware. I use Ender3S1-F4-UBL build.

I had the same problems with very uneven first layers and an "incorrect" or not working bed level compensation.

I followed this suggestion (https://github.com/mriscoc/Ender3V2S1/issues/614#issuecomment-1372696929). This mod somehow made it better but didnt solved it for me completely. 

During installation of the modded extruder assembly i noticed, the CR-Touch probe was not parallel to the x-axis extrusion, turned out my mounting braket where the extruder is getting attached has an angle error in the 90° bend ( the angle its slighty bigger). I designed some kind of small washer to compensate for this, so the probe and extruder assembly will be in parallel to the  x extrusion. I also measured the real probe offset that turned out to be x-46,0 and y-1,0... 

[image]

[image]

After this fix and bed tramming and new mesh generation I finally got a decent and even first layer.

[image]

If you have the same issue with wrong bending angle on the extruder mounting braket, use the attached stl and scale in Z dimension to desired/needed height.
My was 1,2mm in height.

[u-scheibe.zip](https://github.com/mriscoc/Ender3V2S1/files/10465177/u-scheibe.zip)

torstensp, 2023-02-20:
I was facing the same problems with my s1 and s1 pro and stumbled over a solution on YouTube. Couldn't first believe that this would solve the problem, but it did it for me. It has nothing to do with the firmware or any bug. I had to loosen the roles on the z-axis a bit that they move freely. [https://youtu.be/FjLng4CiktA](url)

mohoo, 2023-02-28:
@wibenjamin I didn't quite get it. Do I put the washer between the custom probe mount and the extruder or between the custom probe mount and the extruder mount? I also measured a Y -1.0 in the custom probe mount instead of 0. I did set the -1 in the probe offset though. However, I still do not have a perfect first layer.

wibenjamin, 2023-03-03:
> @wibenjamin I didn't quite get it. Do I put the washer between the custom probe mount and the extruder or between the custom probe mount and the extruder mount? I also measured a Y -1.0 in the [...]

The washer is supposed to sit between the "L"-shaped mounting bracket that is moving along the extrusion and the extruder assembly... Depending on how skew your extruder is now you have to scale the thickness of the washer so your extruder assembly can be mounted perpendicular to the x-axis extrusion

This skewness if not corrected is causing a position deviation of the applied bed level compensation. With the custom probe mount this problem is already mitigated.
[image]

mriscoc, 2023-03-13:
Closed due to this issue was identified as a mechanical problem not a firmware issue.

closed 2023-03-13 by mriscoc