mriscoc/Ender3V2S1 #615: Ender3v2Neo UBL mesh does not seem to be applied

jonathanysp, 2023-01-04:
After updating to the version 20221222 with UBL I have been unable to get good first layer prints. I used bilinear bed leveling with the previous version with good results but decided to use UBL after it was recommended in the latest release notes.

I have a Ender 3 V2 Neo with CR-touch.

Calibration done after installing UBL firmware:

- Set the probe XY offset to (X: -37.1, Y:-11.4) which I measured myself (would be good to get offsets from other Neo owners to confirm my measurements)
- Maximized and centered mesh area
- Set PLA Preheat before running "Auto-build Mesh"
- Saved mesh to memory (the mesh values looked reasonable)
- Updated start g-code:

```
G29 L0; Load the mesh stored in slot 0  
G29 J; Probe 3 points to tilt mesh if needed 
```

First layer results were not good. It was clear that the left side is too close to the nozzle and the right size is too high. Middle is acceptable but top/bottom-middle is not perfect.

I saw bugs mention issues with CR-Touch but it seems specific to S1 model, not sure if it is also an issue with V2Neo.

[image]

mriscoc, 2023-01-05:
Currently there are not mesh errors for 422/427 boards.

None of these command active the mesh leveling:
```gcode
G29 L0; Load the mesh stored in slot 0  
G29 J; Probe 3 points to tilt mesh if needed 
```
For active the mesh leveling use
```gcode
M420 S1
or
G29 A
```

Also, verify that you have a valid mesh in slot 0

mriscoc, 2023-01-05:
You must this when leveling is active:

https://user-images.githubusercontent.com/2745567/210678467-308f3f9c-9c05-4ef8-8824-d4e48a072bc0.mp4

jonathanysp, 2023-01-10:
Adding `G29 A` fixed the issue!

closed 2023-01-10 by jonathanysp

samuelT2, 2023-03-01:
Hi @jonathanysp 

May I ask how your start G-code looks like now?

@mriscoc : In "Advanced Settings" -> "Mesh Leveling" there is a flag called "Activate Leveling". Is this flag obsolete in case of ```G29 A```?

Thank you both!

mriscoc, 2023-03-01:
> there is a flag called "Activate Leveling"

Hi, due to the confusion that "Activate Leveling" tool causes I'm planing to remove the menu item in the next firmware version. 

https://github.com/mriscoc/Ender3V2S1/wiki/Mesh-Leveling-Menu

"Activate Leveling" is only a temporary tool to test if leveling is working, leveling is always disabled after a home operation.

samuelT2, 2023-03-01:
Got it @mriscoc , thank you.