mriscoc/Ender3V2S1 #626: Screen issue when starting auto bed leveling via serial (octoprint)

vrooze, 2023-01-06:
When I start a bed leveling procedure from Octoprint via Bed Visualiser one of three things happen:

I get a generic printing screen and no plot/probe values are shown on screen.
I _dont_ get a generic printing screen and the probe values show perfectly - this is the desired result.
But sometimes (seems to happen more since updating to the current UBL release) I get both!

[image]

The fault of this probably lies somewhere in the middle of the firmware  showing both screens at once and how the procedure is started by the serial console.

Starting it via the screen and knob works every time (no printing screen).

for completeness here is my Bed Visualizer G Code:

```
M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
M29 D  F0 ; Turning off bed leveling while probing, if firmware is set (and turn fade off)
G28      ; home all axes

M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
; Start probing
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
G29 S0   ; Save into Slot 0
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T      ; Query the mesh for below command
; End probing

M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M500 ; store mesh in EEPROM
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display
```

mriscoc, 2023-01-07:
Add a C108 on a new line after the G29 P1, you also have an error on the third line (M29 -> G29)
M500 do not store the mesh in the EEPROM for UBL version is the G29 S0 the command who stored the mesh.

vrooze, 2023-01-08:
Hi 
Thanks for the reply.
I wasn't aware of the custom G-Codes in your firmware (but am now!)
The typo will be corrected I'd missed that one.

Finally regarding the M500. I used https://marlinfw.org/docs/gcode/G029-ubl.html as reference and all the examples end with a M500

I've made those changes and can confirm it's currently warming up and the printing screen is no longer there.

Thanks!

closed 2023-01-08 by vrooze