mriscoc/Ender3V2S1 #641: Request ability to perform Manual Mesh from display without BLTouch, like Jyers

Ergonomicmike, 2023-01-12:
First, I'm not a sponsor. So I don't have any right to ask you for anything.

But, second, I would like to try your firmware. But it doesn't work for my application. (It won't compile without errors, as below.)

My application is that I don't have a BLTouch. Nevertheless, I would like to set AUTO_BED_LEVELING_BILINEAR in config.h and be able to manually set a mesh from my Display.

Now [Jyers](https://github.com/LCh-77/Marlin) allows me to trick it by setting config.h to "No Probe" but still allows me to set AUTO_BED_LEVELING_BILINEAR. This allows me to build a Mesh from the Level Menu on the Display, even though I do not have a BLTouch. Even better, Jyers walks me through to each point for bed leveling from the Mesh Menu, where I enter a value, and after that, it moves the head to the next point. And after all the mesh points have been entered, allows me to "Save mesh."

But when I use WinMerge to set the pertinent settings for leveling your config.h to be the same as the Preset that works for me using Jyers, I receive an error that PREHEAT BEFORE LEVELING is undefined. (I've attached my config.h that works for Jyers.) 

I know that you say to use Octoprint with your Manual Mesh build. But I don't use Octoprint, Nor do I have to with this trick in Jyers.

Since it doesn't appear that branches of Jyers are going to offer MPC, or the new Input Shaping, I would like to try your latest firmware. It would be a nice feature to have the same trick in your Ender3 V2/S1 Professional Firmware that Jyers offers for non-BL Touch users so that we don't have to use Octoprint.

[Configuration.h.zip](https://github.com/mriscoc/Ender3V2S1/files/10398680/Configuration.h.zip)

mriscoc, 2023-01-12:
Use the Manual Mesh (MM) version.
For compile your firmware check https://github.com/mriscoc/Special_Configurations

closed 2023-01-12 by mriscoc

Ergonomicmike, 2023-01-12:
I did compile it. It failed to compile with the changes that I made. But I will try again using your MM Special Configuration.

Thanks.

Ergonomicmike, 2023-01-12:
Okay, I compiled it again using your MM Special Version config.h and config_adv.h

I receive similar errors as before.
```

[{
	"resource": "/c:/Mriscoc.Jan.2023/Ender3V2S1-Ender3V2S1-Released/Marlin/src/lcd/e3v2/proui/dwin.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'struct HMI_data_t' has no member named 'BedLevT'",
	"startLineNumber": 1698,
	"startColumn": 14,
	"endLineNumber": 1698,
	"endColumn": 14
},{
	"resource": "/c:/Mriscoc.Jan.2023/Ender3V2S1-Ender3V2S1-Released/Marlin/src/lcd/e3v2/proui/dwin.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'MSG_PREHEAT_5' is not a member of 'Language_en'; did you mean 'MSG_PREHEAT_4'?",
	"startLineNumber": 2771,
	"startColumn": 59,
	"endLineNumber": 2771,
	"endColumn": 59
},{
	"resource": "/c:/Mriscoc.Jan.2023/Ender3V2S1-Ender3V2S1-Released/Marlin/src/lcd/e3v2/proui/dwin.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'MSG_PREHEAT_5' is not a member of 'Language_en'; did you mean 'MSG_PREHEAT_4'?",
	"startLineNumber": 3292,
	"startColumn": 62,
	"endLineNumber": 3292,
	"endColumn": 62
},{
	"resource": "/c:/Mriscoc.Jan.2023/Ender3V2S1-Ender3V2S1-Released/Marlin/src/lcd/e3v2/proui/dwin.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'MSG_PREHEAT_5_SETTINGS' is not a member of 'Language_en'; did you mean 'MSG_PREHEAT_4_SETTINGS'?",
	"startLineNumber": 3360,
	"startColumn": 47,
	"endLineNumber": 3360,
	"endColumn": 47
},{
	"resource": "/c:/Mriscoc.Jan.2023/Ender3V2S1-Ender3V2S1-Released/Marlin/src/lcd/e3v2/proui/dwin.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'MSG_PREHEAT_5_SETTINGS' is not a member of 'Language_en'; did you mean 'MSG_PREHEAT_4_SETTINGS'?",
	"startLineNumber": 3380,
	"startColumn": 65,
	"endLineNumber": 3380,
	"endColumn": 65
}]
```