mriscoc/Ender3V2S1 #647: Ender 3V2 Bed Visualizer Plugin gcode from wiki does not work

knuxyl, 2023-01-13:
This bug report was closed saying to use the wiki gcode, which is exactly what is described in this post, so the problem still exists. I do not know why this post was sufficient to close this bug report, the advise was not needed because it was exactly what was done.

> Check the Wiki and use the g-code sniped according to your leveling version: non UBL or UBL.

The gcode for the bed visualizer plugin does not create nor view the current mesh. All it does is home x,y, and z. and after a while, sends a completely flat bed mesh to octoprint.
I waited to see if it was waiting for bed termperature or something but it didn't set anything and nothing was increasing.
I followed the entire wiki and setup everything as I was told. I have a Ender 3V2 with BLTouch and v4.2.2 w/Dacai screen.

This was tried multiple times after bed tramming and z probe offset setting

This gcode does not initiate bed level mesh creation. I do not use UBL, so this is the code I need to use, which does not work.

**To take a new mesh in non UBL versions, use this code in the configuration page of the plugin:**
>     M155 S30 ; set temperature reporting delay, use a value longer than the time it takes for your leveling command to complete.
>     G28 XYO ; Do an optional XY home
>     G28 Z ; Do a Z home
>     @BEDLEVELVISUALIZER ; instruct plugin to start recording responses from printer.
>     G29 ; Take a ABL bed leveling mesh.
>     M155 S3 ; set the temperature reporting delay back to a shorter time span.

mriscoc, 2023-01-14:
> This bug report was closed saying to use the wiki gcode, which is exactly what is described in this post, so the problem still exists. I do not know why this post was sufficient to close this bug report, the advise was not needed because it was exactly what was done.
> 
> > Check the Wiki and use the g-code sniped according to your leveling version: non UBL or UBL.
> 
> Update : I changed my Prusa start code to match what is on this github's wiki to create a new mesh before each print and it doesn't work. It appears the G29 gcode is not starting the creation of a new mesh.
> 
> The gcode for the bed visualizer plugin does not create nor view the current mesh. All it does is home x,y, and z. and after a while, sends a completely flat bed mesh to octoprint. I waited to see if it was waiting for bed termperature or something but it didn't set anything and nothing was increasing. I followed the entire wiki and setup everything as I was told. I have a Ender 3V2 with BLTouch and v4.2.2 w/Dacai screen.
> 
> This was tried multiple times after bed tramming and z probe offset setting
> 
> This gcode does not initiate bed level mesh creation. I do not use UBL, so this is the code I need to use, which does not work.
> 
> Per https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint **To take a new mesh in non UBL versions, use this code in the configuration page of the plugin:**
> 
> > ```
> > M155 S30 ; set temperature reporting delay, use a value longer than the time it takes for your leveling command to complete.
> > G28 XYO ; Do an optional XY home
> > G28 Z ; Do a Z home
> > @BEDLEVELVISUALIZER ; instruct plugin to start recording responses from printer.
> > G29 ; Take a ABL bed leveling mesh.
> > M155 S3 ; set the temperature reporting delay back to a shorter time span.
> > ```
> 
> [image]

[image]

knuxyl, 2023-01-14:
I am not using UBL?