mriscoc/Ender3V2S1 #648: Missing sound option

matwalshza, 2023-01-14:
**Describe the bug**
After upgrading to the latest build for S1 F4 UBL I am missing the option in the advanced menu to disable sound.

**Expected behavior**
Advanced menu should have a toggle for sound.

**Version (please complete the following information):**
Ender3S1-F4-UBL-20221222.bin

**Additional context**
Check advanced, config and prepare menu to see if it moved but it has not.
Is there a custom code that can be send with boot config to disable the sound?

mriscoc, 2023-01-14:
To make room in the program memory for the UBL firmware version for the Ender 3S1 F4, it was necessary to disable certain features:
BAUD_RATE_GCODE
INDIVIDUAL_AXIS_HOMING_SUBMENU
SD_PROCEDURE_DEPTH
SDCARD_READONLY (was enabled)
**SOUND_MENU_ITEM**
LONG_FILENAME_WRITE_SUPPORT
BINARY_FILE_TRANSFER
POWER_LOSS_RECOVERY

mriscoc, 2023-01-14:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/502

closed 2023-01-14 by mriscoc

TheWebMachine, 2023-01-14:
> To make room in the program memory for the UBL firmware version for the Ender 3S1 F4, it was necessary to disable certain features: BAUD_RATE_GCODE INDIVIDUAL_AXIS_HOMING_SUBMENU **SOUND_MENU_ITEM** LONG_FILENAME_WRITE_SUPPORT BINARY_FILE_TRANSFER POWER_LOSS_RECOVERY

Ah! Now I know why I couldn't connect Octoprint's firmware updater plugin. haha Do you list somewhere handy which things you had to remove from which firmwares to make room for other things? Like a comparison matrix of the precompiled files? Kinda like what you just mentioned above; what had to be removed to support whatever was added. I had no idea UBL on the F4 came at the cost of PLR, BFT, and the rest above.