mriscoc/Ender3V2S1 #651: building a mesh problem

xx1180, 2023-01-14:
I have installed this on my new S1 plus by changing the screen from touch to knob. i have adjusted the bed size to 300 x 300 but when it probes the bed it only probes 220 x 220 of the bed. i have this working perfectly on my other S1 which is standard 220 x220. can someone please tell me how to probe the full 300mm bed

mriscoc, 2023-01-14:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

xx1180, 2023-01-14:
thanks all sorted now

TheWebMachine, 2023-01-15:
As an aside, you may want to use the CR probe backplate mod mentioned here, as I absolutely had to use it on my Plus in order to get even workable/consistent/reliable mesh results on mine.

https://github.com/mriscoc/Ender3V2S1/issues/614#issuecomment-1374667776

On my Plus, the offsets worked out to be a little different than the other E3 models mentioned in the issue and on Printables. I measured -45.4mmX and -1.8mmY when I installed it and my meshes come out flawless now. I've even been using HS mode with 960 feedrate without more than a 0.0017 deviation in the M48 test.

closed 2023-01-16 by mriscoc