mriscoc/Ender3V2S1 #652: increase temp to 300c

xx1180, 2023-01-14:
I have a s1, s1pro and s1 plus i have this firmware installed on all three machines and now all working as i want except for the fact that the s1 and s1 plus will only go to 260c but the pro goes to 300c. all three printers have the high temp sprite extruders. i did the s1 pro some time ago and can not remember how i got it to go to 300c with this firmware can you please help

AsterDaishi, 2023-01-15:
You'll need to compile your own version of the firwmare to do what you're trying to achieve. Once you load up the firmare source in the editor you have to open your configuration.h file and look for the line #define HEATER_0_MAXTEMP 275
275 here is the max temperature allowed. Keep in mind that the max temperature allowed includes the overshoot. the actual max temperature you're allowed to set is given by the value of HEATER_0_MAXTEMP minus the overshoot value given in #define HOTEND_OVERSHOOT.
In practice this means that for you to be able to set your temperature to 300c you'll need to have your HEATER_0_MAXTEMP set to 300 plus the value of HOTEND_OVERSHOOT.

for example:
#define HEATER_0_MAXTEMP 315
#define HOTEND_OVERSHOOT 15

Keep in mind however that this will mean that your hotend will be allowed to potentially overshoot all the way up to 315 degrees. make sure every component in your hot end assembly is able to withstand the overshoot temperature.

mriscoc, 2023-01-16:
Currently, for printers with the Sprite extruder Pro you should use the special T13 version from here: https://github.com/mriscoc/Special_Configurations/releases/tag/T13 that version has the correct thermistor calibration table for that extruder.