mriscoc/Ender3V2S1 #654: Multiple definitions of run_z_probe and smart_fill_mesh when ProUIex is Disabled

classicrocker883, 2023-01-15:
**Describe the bug**
Error after compiling
>Marlin/lib/proui/libproui.a(proui.cpp.o): In function `Probe::run_z_probe(bool)`:
proui.cpp:(.text._ZN5Probe11run_z_probeEb+0x0): multiple definition of `Probe::run_z_probe(bool)`
.pio/build/GD32F103RE_voxelab_maple/src/src/module/probe.cpp.o:/home/andrewleduc88/PlatformIO/Mriscoc-2023/Marlin/src/module/../inc/../core/macros.h:436: first defined here
Marlin/lib/proui/libproui.a(proui.cpp.o): In function `unified_bed_leveling::smart_fill_mesh()`:
proui.cpp:(.text._ZN20unified_bed_leveling15smart_fill_meshEv+0x0): multiple definition of `unified_bed_leveling::smart_fill_mesh()`
.pio/build/GD32F103RE_voxelab_maple/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o:/home/andrewleduc88/PlatformIO/Mriscoc-2023/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1437: first defined here

**To Reproduce**
Steps to reproduce the behavior:
1. Disable `ProUIex` (comment out `#define ProUIex 1` in Configuration.h)
2. enable UBL auto bed leveling

**Expected behavior**
It should compile without error.

**Additional context**
Upon further discovery, in file **src/feature/bedlevel/ubl/ubl_G29.cpp** @ line 1434 `#if DISABLED(ProUIex)` and `void unified_bed_leveling::smart_fill_mesh()`

it seems that when ProUIex is disabled, this definition of smart_fill_mesh() is defined in two places, here and in the lib/proui/libproui.a or similar libproui_xx.a files.

mriscoc, 2023-01-16:
I was not able to replicate, UBL is enable. Maybe I have that already fixed in my current sources.

[image]

classicrocker883, 2023-01-17:
BLTOUCH must also be enabled. using the Ender3V2-422-BLT configurations, the error still happens.

classicrocker883, 2023-01-17:
I find out it is because I had `#define HAS_GCODE_PREVIEW 1` enabled, because this worked in the past having ProUIex disabled. I do not know how `run_z_probe()` and `smart_fill_mesh()` can be related to the model preview for SD card.

is there a way so ProUIex can be disabled and the GCODE_PREVIEW work like it was able to before?

classicrocker883, 2023-01-17:
in file **src/lcd/e3v2/proui/dwin_defines.h** --> line 77 under 
>// Only for Professional Firmware UI extensions
#if ENABLED(HAS_GCODE_PREVIEW) && DISABLED(ProUIex)
  //#error "HAS_GCODE_PREVIEW requires ProUIex."
#endif

After I commented out the error so HAS_GCODE_PREVIEW can be enabled,

I was able to remove/comment out the whole `float Probe::run_z_probe(const bool sanity_check/*=true*/) { ... }`  from file **src/module/probe.cpp** --> line 687, from `#if ProUIex` to `#endif`

and `void unified_bed_leveling::smart_fill_mesh() { ... }` from file **src/feature/bedlevel/ubl/ubl_G29.cpp** --> line 1434, from `#if ENABLED(ProUIex)` to `#endif`

I havent yet tested the firmware to see how it works, but it compiled successfully. to reiterate, I had UBL and BLTOUCH enabled like in the Ender3V2 configuration files, but `ProUIex` /  `HAS_TOOLBAR` disabled/commented out.

this is so the SD card print preview can be viewed while `ProUIex` and Toolbar is disabled.

mriscoc, 2023-01-18:
I think that you need ProUIex to work with the new G-code preview library.

If you want to disable ProUIex maybe could be better to work with the ProUI from the Marlin repo: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/e3v2/proui
I wrote and post that UI specially to do not to need ProUIex.

closed 2023-03-13 by mriscoc