mriscoc/Ender3V2S1 #664: Toolbar not being drawn

robertwhitlock3, 2023-01-19:
Hi mriscoc,
  I've really enjoyed configuring and getting your firmware suited to my needs - I recently put a SKR Mini e3 v3 into my Ender 3 v2 Neo.
  
  One niggle I noticed was that scrolling right on the (ProUI) LCD main menu made the cursor disappear. Thought this was odd and started diving into the source.
Discovered it was because of the toolbar. Cool feature! But on my LCD screen there's just a blank line where the toolbar items should be drawn. 
I had a play in the Toolbar config menu, everything in there displays, and looks ok, but it still won't draw the toolbar on the main menu.
   
   So I started adding some debug, thinking it was because the menu wasn't being built correctly. But `Draw_ToolBar` Was correctly getting a `TBItem` with a non-zero `icon`, then I thought `MENU_ITEM_F` might be failing, but that returned a valid pointer.

  Now I'm thinking it's something to do with the `ToolBar.draw()` call since I never see any debug in the `onDrawTBItem` method. I can see in `ToolBarClass`, there's an override specified for `draw`, but can't find the implementation. Debug wasn't printed for the draw method in the base class (`MenuClass`). Pulling out a code snippet to try to understand the inheritance into a new VS C++ project doesn't build, since ToolBarClass doesn't implement its override.
  
 Am I missing something obvious? I've seen the demos you've posted so I know it can work, but I'm not seeing the same.

mriscoc, 2023-01-20:
Duplicate: https://github.com/mriscoc/Ender3V2S1/issues/249
Duplicate: https://github.com/mriscoc/Ender3V2S1/issues/478

closed 2023-01-20 by mriscoc

robertwhitlock3, 2023-01-20:
Many thanks and makes sense.
I was adding my initials to the end of the author so I can tell if it's using my config. Will make these changes!