mriscoc/Ender3V2S1 #669: LEVELING_BED_TEMP variable can not be used

2probro, 2023-01-23:
**Describe the bug**
The bug is not in the official marlin 2.1.x bugfixes folder but I can find him here. If you try to compile marlin with your marlin-folder you get an error:

```
In file included from Marlin\src\feature\../module/probe.h:33,
                 from Marlin\src\feature\bltouch.cpp:39:
Marlin\src\feature\../module/../lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\feature\../lcd/marlinui.h:49,
                 from Marlin\src\feature\cancel_object.cpp:29:
Marlin\src\feature\../lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/marlinui.h:49,
                 from Marlin\src\MarlinCore.cpp:57:
Marlin\src\lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
```

This error occurs because the variable "LEVELING_BED_TEMP" doesn't exist if disable Preheat before levelling ==> //#define PREHEAT_BEFORE_LEVELING

You should implement a way to only use the "LEVELING_BED_TEMP"-variable when "PREHEAT_BEFORE_LEVELING" is enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn off PREHEAT_BEFORE_LEVELING
2. Try to compile
3. Get the error `'LEVELING_BED_TEMP' was not declared in this scope`

**Expected behavior**
LEVELING_BED_TEMP-variable shouldn't be used cause PREHEAT_BEFORE_LEVELING is turned off

**Version (please complete the following information):**
2.1.x bugfixes, I used the marlin source code zip provided by you. Downloaded today.

mriscoc, 2023-01-24:
Change in dwin.h:
```cpp
int16_t BedLevT = LEVELING_BED_TEMP;

to:

int16_t BedLevT;
```

mriscoc, 2023-03-13:
Fixed in latest version.

closed 2023-03-13 by mriscoc