mriscoc/Ender3V2S1 #670: Can't compile becasue of BedLevT

2probro, 2023-01-23:
**Describe the bug**
I tried to compile and I get an error:

```
Marlin\src\gcode\bedlevel\abl\G29.cpp: In static member function 'static bool GcodeSuite::G29()':
Marlin\src\gcode\bedlevel\abl\G29.cpp:453:22: error: 'struct HMI_data_t' has no member named 'BedLevT'
  453 |             HMI_data.BedLevT
      |                      ^~~~~~~
```

I have enabled "PREHEAT_BEFORE_LEVELING" and I get this error, if I disable "PREHEAT_BEFORE_LEVELING" there is no error.
So there is a problem with this feature.

I also recently found another bug with PREHEAT_BEFORE_PROBING, but this one is different, so I decided to make a separate bug report for it. 

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on PREHEAT_BEFORE_LEVELING
2. Compile
3. See error

**Expected behavior**
It should just compile but I get an error

**Version (please complete the following information):**
Latest Marlin 2.1.x (downloaded today from https://github.com/mriscoc/Ender3V2S1)

classicrocker883, 2023-01-24:
in file **src/lcd/e3v2/proui/dwin.h** line 113
look for `int16_t BedLevT = LEVELING_BED_TEMP;`

add 
>#if ENABLED(PREHEAT_BEFORE_LEVELING)
int16_t BedLevT = LEVELING_BED_TEMP;
#endif

mriscoc, 2023-01-24:
That error is because you removed the definition of BedLevT in dwin.h

mriscoc, 2023-01-24:
> #if ENABLED(PREHEAT_BEFORE_LEVELING)
> int16_t BedLevT = LEVELING_BED_TEMP;
> #endif

Due to ProUIex library is not good idea to change the structure of HMI_data.

closed 2023-01-24 by mriscoc