mriscoc/Ender3V2S1 #676: Thermistor 66 Mintemp error

dylanmaloney, 2023-01-25:
ust upgraded to Thermistor 66. Newly built marlin that I'm sure needs some work. Anytime I run a PID tune or MPC I get error MinTemp before it begins heating. I have set it to 30000ms preheat time and 10 low temp errors.

Bug Timeline
 
Started since swapping thermistors and updating configs.

**To Reproduce**
Steps to reproduce the behavior:
#define MILLISECONDS_PREHEAT_TIME 30000
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10
Run MPC or PID through Marlin Professional

**Expected behavior**
Heat upon request

**Version (please complete the following information):**
Firmware version and release date
2.1.3 Most recent, downloaded yesterday. 1/23/23

mriscoc, 2023-01-25:
I'm not sure if the board will support this kind of thermistor because it have a very high impedance. Normal thermistor have 100KΩ but the T66 has 4.7MΩ
Did the vendor of the thermistor test it with a Creality 422/427 board?

dylanmaloney, 2023-01-25:
There are no specific boards listed on the website but searching through reviews I see many E3V2 success stories. None specific on what boards they have though.

mriscoc, 2023-01-26:
I don't have that thermistor to do test. so it is not possible for me to replicate your issue, please try with the latest Marlin bugfix with DWIN_LCD_PROUI enabled and report back your results.

dylanmaloney, 2023-01-26:
No worries. I’ll try and get back to you tonight. Thank you

dylanmaloney, 2023-01-26:
I ran into a common issue I tend to find, maybe you can help. Ender 3 v2 4.2.7 Board Config files. They are never in the updated releases. I bought the V2 stock with the 4.2.7 Board too.

mriscoc, 2023-01-27:
You need to download the Marlin configuration files if you try to compile the Marlin bugfix source.

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x

dylanmaloney, 2023-01-28:
After quite some time I was able to compile a semi-working bugfix/professional firmware. Working: firmware created, boots, heats to 485 centigrade, reads tempurature somewhat accurately etc. 

Doesn't Work: Steppers... Motors? XYZ. 

I compiled it with your config files from your professional firmware and Marlin's Bugfix configs. Was wondering if you could take a peek or maybe have an Idea was causing the motors not to have power. I'm 99% certain its firmware related as thats the only thing I have messed with.
[Configuration.zip](https://github.com/mriscoc/Ender3V2S1/files/10528488/Configuration.zip)

mriscoc, 2023-01-30:
Check the steppers driver settings.

mriscoc, 2023-03-13:
No more feedback

closed 2023-03-13 by mriscoc