mriscoc/Ender3V2S1 #679: Ender 3v2 restart

Slavikpd, 2023-01-25:
Good day. I have an Ender 3v2 main board 4.2.2. After updating the latest firmware version Ender3V2-422-BLTUBL-MPC-20221222.bin
it happens to me that the printer restarts itself before printing. It also happened that the picture fell apart and I had to turn it off. And it also happens that some layers are damaged. Can you advise me where the problem might be. Well thank you.

[image]

giacomoferretti, 2023-02-11:
Also happening to me. If I turn on the printer with the SD card inserted, the printer will restart until I remove the SD card.

Slavikpd, 2023-02-11:
ThankYou

giacomoferretti, 2023-02-13:
> > Also happening to me. If I turn on the printer with the SD card inserted, the printer will restart until I remove the SD card.
> 
> Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/480

Thanks, didn't see that.

J-oe-rg, 2023-02-26:
Hi Miguel, I have the same issue reported in https://github.com/mriscoc/Ender3V2S1/issues/480 's first posting.

Original Creality 8GB Card formatted with FAT32, 4096 byte cluster size.
Doesn't matter if there are files on the card or not. Media update is off.

For me that issue did not occur before on stock or Jyers firmware. (There was an assumption by another user.)

By the way, last firmware update was made with this card but I formated it afterwards.
A second card 16GB Samsung doesn't have that issue. Next release I will update with that card to see if there is some coincidence.

Printer: Ender 3 V2, board 4.2.2
Firmware: Ender3V2-422-MM-MPC-20221222.bin

**Workaround:** Just take out the card for some seconds and put it in again. Boot loop stops.

Thank you for the great firmware!

mriscoc, 2023-03-13:
Please test with latest version.

J-N-R, 2023-03-30:
I'm getting it too. Frozen on startup

Printer: Ender 3 V2, board 4.2.2
Firmware: Ender3V2-422-BLTUBL-MPC-20230312.bin

mriscoc, 2023-03-31:
Closed as duplicate

closed 2023-03-31 by mriscoc

J-N-R, 2023-04-03:
Any update on this? All the duplicate bugs are closed but its still freezing using the given 8gb card (FAT32 formatted)

mriscoc, 2023-04-08:
Informative link: https://www.3dprintgorilla.com/ender-3-pro-v2-neo-s1-sd-card-size