mriscoc/Ender3V2S1 #683: M117 Messages Not Showing While Printing, Instead Printing Object 0 is displayed

xantrk, 2023-01-26:
[image]
My gcode has M117 commands after each layer to tell me current layer and no of layers as in:
M117 Printing Layer {layer_num+1}/[total_layer_count]

For the first time I started using this firmware today so no idea if it is a recent bug. I am using self compiled Ender3v2, 4.2.2 UBL BLTouch, LA version. 

Thanks a lot for your efforts!

mriscoc, 2023-01-27:
You may have another plugin writting to the screen.

TheWebMachine, 2023-01-27:
I was thinking the Cancel Objects plugin in Octoprint but OP didn't specify if they were using SD card solo or connected to Octoprint.

xantrk, 2023-01-27:
> 

> You may have another plugin writting to the screen.

> I was thinking the Cancel Objects plugin in Octoprint but OP didn't specify if they were using SD card solo or connected to Octoprint.

I am printing directly from SD card. Searching through gcode manually, I don't see anything with "printing object" in there, which is quite weird

mriscoc, 2023-01-27:
Are you using a precompiled firmware or a custom compile?

xantrk, 2023-01-27:
Custom compile 99.9% based on 4.2.2 UBL BLTouch, LA.
Just modified a few offsets etc. i can share my configs in a few hours when I'm back near printer.

mriscoc, 2023-01-27:
> Custom compile 99.9% based on 4.2.2 UBL BLTouch, LA. Just modified a few offsets etc. i can share my configs in a few hours when I'm back near printer.

Try with the precompiled firmware versions.

xantrk, 2023-01-28:
> You may have another plugin writting to the screen.

Turns out it was PrusaSlicers M486 S0 command stating with object (index) is being printed. This setting was turned on for ages for me, not sure if it is by design for printer to ignore subsequent M117 messages after M486.

closed 2023-01-28 by xantrk

TheWebMachine, 2023-01-28:
If I'm not mistaken, I ran into issues in the past with M117 messages in other Marlin-based firmware (Prusa comes to mind) if you include certain chars in the message; like if you try to use double-quotes around the message, some firmware will ignore the M117 and not update the display. I wonder if any such bugs still exist upstream in Marlin or have found their way into this build.

That said, I use Prusaslicer with verbose object gcode, too, (and the Cancel Objects plugin in Octoprint) and have yet to see an M486 prevent future M117 messages from showing. So, not sure why it happened to you. Maybe a back-to-back M486 then M117 results in the M117 not getting displayed when printing from SD or something like that.