mriscoc/Ender3V2S1 #684: The flow rate does not change

elVucko, 2023-01-26:
Hello,
I have Ender 3v2 with motherboard 4.2.2. and the latest firmware ( Ender3V2-422-BLTUBL-MPC-20221222.bin ) is installed.
The problem I have is that the printer does not change the flow rate even though Cura is set to change the flow after the first layer.
Also when I try to print Flow Tower flow remains unchanged. It can only be changed manually during printing in the Tune menu.

mriscoc, 2023-01-30:
Changes in the flow rate are not immediate and depend on the number of moves currently in the queue. Look for https://marlinfw.org/docs/gcode/M221.html in your G-code file.

Tested and M221 does indeed change the flow value.

elVucko, 2023-02-09:
M221 changes the flow value, but the problem is when I set it in Cura. Then nothing happens and it has to be changed manually or using the M221. But it doesn't matter.

mriscoc, 2023-02-09:
> but the problem is when I set it in Cura

Check in Cura forums. 3D printers use G-code commands.

closed 2023-02-09 by mriscoc