mriscoc/Ender3V2S1 #686: M503 dumps settings in a bad ordering

dfinberg, 2023-01-26:
This is somewhere between a feature request and a bug, but it's annoying and can screw up your config if you don't catch it, and the fix takes seconds.

**Describe the bug**
The M503 dump code for custom gcodes calls their print methods in numerical order via custom_gcode_report(). The problem with this is that the dimension gcodes (C100-102) are numerically after the probing gcode (C29), so if your bed is larger than the default dimensions, it will clamp those values to the default bed if you feed them back in the same order that M503 produces them.

**To Reproduce**
Steps to reproduce the behavior:
1. Dump a config via M503/Octoprint and save the settings into a config file
2. Reflash firmware and load that saved config file
3. Check the mesh inset dimensions, and it is not what was set via C29, since the bed was too small at that point.

**Expected behavior**
Just move the C29 output to below the C100-2 codes

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
12/22/22 release

**Additional context**
Add any other context about the problem here.

mriscoc, 2023-02-01:
Hi, thanks for your suggestion, I will move the C29 after the C10X custom G-codes. but maybe this could be useful for you: https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files

mriscoc, 2023-03-13:
Fixed in latest version

closed 2023-03-13 by mriscoc