mriscoc/Ender3V2S1 #694: Ender 3 v2 neo based on GD32F303 support

vfiuchcikicshuusrch, 2023-01-28:
Can it work with Ender 3 v2 neo based on GD32F303 chip ?

mveplus, 2023-01-29:
Hi @vfiuchcikicshuusrch, 
Good question, I tried all these and it's stuck on the printer gets stuck on the Creality display, but I'm not running ender vanilla flash I'm with E3V2-BLTouch-3x3-v4.2.2-v2.0.1.bin. To fix I need to reflash back with it. 
`
Ender3V2-422-BLTUBL-MPC-20221222.bin
Ender3V2-422-MM-MPC-20221222.bin
Ender3V2-SKRME3V3-BLTUBL-MPC-20221222.bin
`
Renaming the files and trying to multiply tiles did not help. N.B. My Ender3v2 is with board version 4.2.2 and SoC GD32F303, added CR touch and a filament sensor, not other mods.

vfiuchcikicshuusrch, 2023-01-29:
[GD-Ender-3 V2-Marlin2.0.8.2-HW-V4.2.2-SW-V1.0.4_E_N_20211230.zip](https://github.com/mriscoc/Ender3V2S1/files/10530496/GD-Ender-3.V2-Marlin2.0.8.2-HW-V4.2.2-SW-V1.0.4_E_N_20211230.zip)

Maybe this one can help you ?