mriscoc/Ender3V2S1 #695: Filament doesn't feed after resuming from smart pause

sammcj, 2023-01-28:
**Describe the bug**

There is a bug in the upstream Marlin firmware that causes the printer to fail to extrude/feed filament after resuming from a (smart) pause.

This is a real problem when using OctoPrintAnywhere (or similar) as if they detect a potential print issue and pause the print - your whole print job is ruined as you have to cancel your print.

- Marlin bug: https://github.com/MarlinFirmware/Marlin/issues/24996
- Workaround / patch: https://github.com/MarlinFirmware/Marlin/issues/24996#issuecomment-1356329828

**To Reproduce**
Steps to reproduce the behaviour:
1. Print (I'm using an Ender 3 v2 Neo)
2. Issue a smart pause (M600) OR issue a pause from OctoPrintAnywhere
3. Resume printing
4. Note that while the head and bed move - no filament is extruded

**Expected behaviour**

Resuming from a Pause (M600 or otherwise) should continue printing.

**Version (please complete the following information):**

- 20221222 - Dec 22, 2022

TheWebMachine, 2023-01-29:
I have been able to use M600 for filament change (layer color changes) without issue on my E3 Plus (F4 board) with this firmware. I don't experience any extruder issues with this command.

I *did* however find that the default Ender 3 profile for OctoLapse incorrectly has the G90/G91 Affects Extruder option set to False, which caused issues with the extruder for me as soon as the print progressed to the second layer because OctoLapse was switching the extruder to Absolute mode. This has been the only issue I've encountered that caused the extruder "stall" you describe.

Which board does your Neo have? Have you confirmed other plugins like OctoLapse or OctoPrintAnywhere aren't inserting G90/G91 commands during the print or pause? Absolute mode on the Extruder will make it appear to stall or jitter if the E moves are relative in gcode. I figured out the OctoLapse issue I ran into by downloading and comparing the OctoPrint Serial Log with my print gcode.

sammcj, 2023-01-29:
Hey @TheWebMachine, my Neo has:

- The newer revision 4.2.2 board
- No hardware / extruder mods
- OctoLapse is disabled
- OctoPrintEverywhere doesn't have any settings for the pause command, I'm not quite sure how it works actually! 

The only G90 and G91 I can see is in OctoPrint after a print is _cancelled_ which I'm assuming is ok.

[image]

MarkforU, 2023-01-29:
I have never used the Smart Pause through the OctoEverywhere, only reason I don't use smart pause, is because of all the complaints I've seen, which is probably a Marlin issue, but I've heard of a lot of nightmares on Smart Pause...such as after resuming it goes to the beginning of next level, or it doesn't continue from the point that it was paused. I've always advised filament change to do any type of pausing. This has even helped me on a clog issue that I had to change the nozzle out and went back to perfect printing.

sammcj, 2023-01-29:
I find it super useful, it's caught a few spaghetti disasters before they wasted a lot of filament or worse yet - caused a fire / damage to the printer.

MarkforU, 2023-01-30:
I let Gadget catch it and notify me where I can take a look. I was printing Crystal Dragons and Gadget would go off on every print detecting lift, where there wasn't a lift but the design. 
So, myself, I like the alerts, but don't want it to stop because it thinks there's a lift and there isn't. On long prints though, I understand. Usually Gadget would be blowing up my phone in those case's, but I still have it report, but not stop or pause, I take a look at my phone and decide if ot needs canceled or keep going.

sammcj, 2023-02-11:
Here's bigtrees patch for their FW as an example - https://github.com/deveth0/Bigtree-SKR-Mini-v3-Ender3v2-Marlin/commit/8a096b1de2c49506a14aed789a38c8505eb1e74d#diff-9876938bb66214705f89adf5ab50fd49d5504d318662b5c996ac0eba2c434587

mriscoc, 2023-02-13:
Pause/Resume works as intended, but there are reports of problems but totally different ones, someone reports over extrusion, others no extrusion, etc. I think all the issues are related to how the slicer is preparing the pause/resume operation, for this firmware (compatible with Marlin 2 G-codes) only one M600 is needed to initiate an advanced pause. The firmware will take care of all operations.

mriscoc, 2023-03-13:
Issue can't be replicated, possibly related to slicer/host configuration.

closed 2023-03-13 by mriscoc