mriscoc/Ender3V2S1 #698: homing after power out

liamwerk7676, 2023-01-29:
**Describe the bug**
power went out it wanted to resume and it preheated and then gone to the point of the print and started to auto home a normal auto home not a xy auto home it homed z 2

**To Reproduce**
Steps to reproduce the behavior:
1.power out
2.continue

**Expected behavior**
home x and y and continue print

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date
20221222
**Additional context**
print took a loong time so this was terible and used up a lot of filament

mriscoc, 2023-01-29:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/571

liamwerk7676, 2023-01-29:
no it literly homed it dit not continue printing

mriscoc, 2023-01-30:
Closed because it is a duplicate of https://github.com/mriscoc/Ender3V2S1/issues/571

closed 2023-01-30 by mriscoc