mriscoc/Ender3V2S1 #703: 300C option for upgraded ender3 S1

Nathan-Holt, 2023-01-31:
I have an upgraded nozzle on my S1, and would like to be able to use both this software, and also higher temp materials such as PC.  Currently the only way is through Creality official software. 

 Adding a 300C option to the firmware download would allow for those with the correct upgrades to utilize them fully.

mriscoc, 2023-01-31:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/652

xx1180, 2023-01-31:
you need more than the nozzle to print at 300c you also need the upgraded heatbreak as the standard heatbreak has a piece of Teflon tube inside and you will also need the uprated thermistor and heater cartridge

xx1180, 2023-01-31:
i have my bin file for the s1 if you would like i could send it to you

mriscoc, 2023-01-31:
Closed as duplicate of#652

closed 2023-01-31 by mriscoc

Nathan-Holt, 2023-01-31:
> i have my bin file for the s1 if you would like i could send it to you

that would be awesome! i do have the full heatbreak assembly. just mistyped it as nozzle as im a bit lazy about my terminology.

xx1180, 2023-02-01:
sorry for delay just been busy

> > i have my bin file for the s1 if you would like i could send it to you
> 
> chip type requires i use an F4 bin that would be awesome! i do have the full heatbreak assembly. just mistyped it as nozzle as im a bit lazy about my terminology.

need an email address to send you file the site won't let me paste it here

xx1180, 2023-02-01:
sorted it out just started my own repository just go here for your file
https://github.com/xx1180/S1-F4-T13

Nathan-Holt, 2023-02-05:
page isnt there?

xx1180, 2023-02-05:
please let me have your email address and i can send it to you. don't know why you can not see it i have just looked and it is there when i click on it

Nathan-Holt, 2023-02-05:
***@***.***

xx1180, 2023-02-05:
looks like system won't let you send email as address was starred out try sending again but replace @ with AT

xx1180, 2023-02-05:
try downloading again also as i just realised the i had my page set as private and have changed it to public
https://github.com/xx1180/S1-F4-T13

Nathan-Holt, 2023-02-05:
nathandholt at gmail dot com