mriscoc/Ender3V2S1 #716: Tramming/Mesh probe postions

YanivKrit, 2023-02-03:
**Describe the bug**
I am using fw [Ender3S1-F4-UBL-20221222.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20221222/Ender3S1-F4-UBL-20221222.bin)
On "Tramming Wizard" when I run each corner it seems like there is a shift from where the probe "samples" the Z height and where I think it should sample, when choosing each corner the probe goes to positions (subtract the offset from the real position):
FL = 10,10
FR = 175,10 (very far from the right)
BL = 10,179.5 (very far from the back)
BR = 175,179.5 (vart far from the back and from the right)

but when activating "Manual Tramming" it goes to positions (real position):
FL = 30,30
FR = 190,30
BL = 30,190
BR = 190,190

when creating a mesh the center point (and all the points) is shifted and istead of being at 110,110 its at 92.5,94.75 and from attached fustion sketch you can see that the difference is 17.5,15.25 that if we add this offsets to the 4 corners of the tramming wizard then we will recieve positions that make more sense
FL = 27.5,27.5
FR = 27.5,192.5
BR = 192.5,194.75
BL = 27.5,194.75

BUT when editing the center point of the mesh then it goes to position 110,110 (index 2,2 for 3x3 mesh)
so I think that somewhere in the code there is a shift of 17.5 for X position and 15.25 for the Y position when probing or something like that

**Version (please complete the following information):**

[image]

mriscoc, 2023-02-04:
Check:
https://github.com/mriscoc/Ender3V2S1/wiki/Bed-tramming-menu#note2
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

YanivKrit, 2023-02-04:
I understand now that you are the admin that replyed to my post on facebook, did the calibratio guide again as you told me and it seems that the problem was with the "mesh insert" thank you!

closed 2023-02-04 by YanivKrit