mriscoc/Ender3V2S1 #718: Mesh leveling doesn't work when 'Activate leveling' is turned on?

sassysander, 2023-02-04:
There seems to be an issue with the mesh leveling when 'Active leveling' is turned on?

If I create a new mesh in the printer menu, either the default 5x5 or any of the other options, the mesh is created and stored in the '0' slot when I press 'Save' after the probing has completed. I can verify this because the mesh it created correctly shows up if I select 'View Mesh', as it displays the values previously probed, so it appears the mesh is correctly stored in the '0' slot.

However for some reason mesh leveling doesn't seem to work when 'Activate leveling' is turned on in the printer menu, as when I start a print with 'Active leveling' turned on the printer doesn't use the stored mesh. I can replicate this by turning the 'Activate leveling' on or off in the printer menu. If I turn it off leveling works, if I turn it on leveling doesn't work, which seems somewhat counterintuitive.

Below is my start up script, I am using the G29 command to activate leveling and load the '0' mesh. Could this issue be due to my use of the G28 and G29 commands? Do these toggle the printer's 'Activate leveling' menu option on and off? That appears to be the case?

; Start up script
G21 ; set units to millimeters
G90 ; use absolute coordinates
G28 ; home XYZ axis end-stops
M104 S{first_layer_temperature[0]} ; set final nozzle temp
M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize
G29 A ; activate bed leveling
G29 L0 ; load first (0) bed leveling mesh
G29 F10 ; set fade height
G0 X0 Y0 F5000 ; move to front
G92 E0 ; zero the extruded length
G0 Z0.4 ; drop Z to 0.4mm from the bed for purge
G1 X50 E25 F500 ; purge 25mm of filament in a 50mm line
G1 X80 F4000 ; quickly wipe away from the purged filament line
G92 E0 ; zero the extruded length
; Start print

Crazy-Charles, 2023-02-04:
Try activating the mesh levelling after loading it and setting fade height.
i.e
G29 L0
G29 F 10.0
G29 A

grrmisfit, 2023-02-05:
Your issue seems to be a common one and an easy fix. It stems from the probe location, using this mod https://www.printables.com/model/257460-ender-3-s1-zero-offset-cr-touch-mount moves it to the back. The reason it works with mesh off is cause your bed is mostly level but the mesh is using bad values so when you turn on mesh it doesn't work right. There's an old issue here that has lots of input on this issue

mriscoc, 2023-02-06:
"Activate leveling" tool function is explained in the Wiki. Also a correct start script is in the Wiki.

closed 2023-02-06 by mriscoc