mriscoc/Ender3V2S1 #721: GD32 chip support on Ender 3 v2 Neo

talentoscope, 2023-02-04:
**Describe the bug**
This firmware does not support the newer GD32 chips in the newer Ender 3 v2 Neo machines (of which I have one)

**Version (please complete the following information):**
Latest

**Additional context**
The board is the 4.2.2 one.

mriscoc, 2023-02-04:
https://github.com/mriscoc/Ender3V2S1/wiki#requirements

closed 2023-02-04 by mriscoc

sammcj, 2023-02-12:
What happens when you try to run it @talentoscope?

I also have a 4.2.2 GD32F303 and it works fine.

[image]

talentoscope, 2023-02-12:
Oh wow, really?

It just hangs at the Creality logo. I'm using the Ender3V2-422-BLTUBL-MPC-20221222.bin ?

Really want to get this working, but it's annoying

sammcj, 2023-02-12:
Yep! No issues at all (other than the known Marlin smart resume/M600 bug #695) and I'm running the same - `Ender3V2-422-BLTUBL-MPC-20221222.bin`

[image]

sammcj, 2023-02-12:
I just installed a Spider hot end today and I'm going to do a custom FW build with some updated settings - I guess I'll find out if it works a second time!

talentoscope, 2023-02-12:
Ahh, it's a GD32F303 here. That might be it. will have to hunt around.

Thank you :)

sammcj, 2023-02-12:
Could be unrelated but I had that with the SD Card that came with the printer - I swapped it out for a 16GB one I had lying around that I formatted to FAT (mbr with 4k blocks).

talentoscope, 2023-02-12:
Doubtful as I can flash the Creality one back afterwards but anything is
with a try

sammcj, 2023-02-12:
The creality one worked on the stock SD card for me.

Make sure you format with a 4096 block size, e.g if you're on macOS:

sudo newfs_msdos -F 32 -b 4096 disk4s1

talentoscope, 2023-02-12:
Got it, thanks Sam, appreciated!

talentoscope, 2023-02-20:
Just thought I'd update here for anyone interested.
Bought a Kingston 32GB Class 10 SD card and I now have this on my Ender 3 v2 Neo with the F303 chip.

Thanks!