mriscoc/Ender3V2S1 #731: Cancelling print Cause Bed to move

JonSnow88, 2023-02-13:
When printing from Octoprint and then cancelling the print using the stock Ender 3 V2 LCD screen, the bed will move completely to the front like as if the print is "completed",  few times if its during the first few layers it hits the bed clips and could easily scratch the bed.  

When the same file is printed from the SD card then I cancel the print from the LCD screen all motors just stop in place. 
When printing from Octoprint and Cancelling from Octoprint all motors just stop in place.

mriscoc, 2023-02-16:
The bed move after cancelling an SD print is coded at:

https://github.com/mriscoc/Ender3V2S1/blob/68791f246b8da6ff3e69b2a88b601d36fd73e9e7/Marlin/src/lcd/e3v2/proui/dwin.cpp#L1640

You can change your Y park point or edit the source code to modify that behavior.

closed 2023-02-16 by mriscoc

JonSnow88, 2023-02-17:
@mriscoc I am not cancelling a SD card print, I don't use the SD card at all. 

This happens when printing from Octoprint and using the Ender 3V2 LCD screen to cancel, your previous FW did not do this.