mriscoc/Ender3V2S1 #734: Horizontal line artefacts with IS / LA?

RoP09, 2023-02-15:
Hello,

with the latest update i noticed some irregular horizontal line artefacts in my prints which i think were not there before.
E.g. especially visible when I print a X Y Z calibration cube but not at all when I print a cylinder in vase/spiral mode.

Could that be due to the new features introduced lately (especially Input Shaping...)?

I am using a BTT SKR E3 V3 board on Ender 3 V2 with both LA and IS enabled.

BR

classicrocker883, 2023-02-15:
I think input shaping is enabled no matter what if it was so when firmware was compiled. I have noticed this too. even if you set the input shaping values to Zero or try to disable, you get artifacts. have you done the calibrations required to use input shaping?

mriscoc, 2023-02-16:
IS or LA are enabled only in [special firmwares with IS or LA](https://github.com/mriscoc/Special_Configurations). Regular releases in this repository don't have IS or LA.

RoP09, 2023-02-16:
@mriscoc : Yes, but it seems the problem is it cannot be disabled even by M593 F0.

> I think input shaping is enabled no matter what if it was so when firmware was compiled. I have noticed this too. even if you set the input shaping values to Zero or try to disable, you get artifacts. have you done the calibrations required to use input shaping?

That's how it looks like, and no I did not do any calibration yet.

mriscoc, 2023-02-22:
Did you check if the latest Marlin bugfix has the same behavior?

closed 2023-05-05 by github-actions[bot]