mriscoc/Ender3V2S1 #738: Freezes on boot

PolyPlugins, 2023-02-16:
**Describe the bug**
After installing the latest version, if we don't interact with the printer when it first boots, it freezes and the credit popup will keep popping up over and over again

**Version (please complete the following information):**
20221222

mriscoc, 2023-02-16:
When checking the Ender3V2-422-BLTUBL-MPC-20221222.bin version, the reported problem does not appear.

PolyPlugins, 2023-02-16:
> When checking the Ender3V2-422-BLTUBL-MPC-20221222.bin version, the reported problem does not appear.

Apologies, we're using Ender3V2-422-MM-MPC-20221222.bin

joost01, 2023-02-28:
I also have this problem with the Ender3V2-422-BLTUBL-MPC-20221222.bin file. Booting without the SD card seems to solve the problem, at least Octoprint can connect to the printer again. With the SD card inserted Octoprint won't connect and every 3-5 seconds the credits pop-up is displayed. Also, you cannot manually change settings on the display.

J-oe-rg, 2023-03-05:
Sounds to me like the issue described in #679 and #480.

mriscoc, 2023-03-13:
The issue can't be replicated in the test group.

closed 2023-03-13 by mriscoc