mriscoc/Ender3V2S1 #743: [BUG] Pause/Resume parks twice, then start with incorrect Z value

tomaszczura, 2023-02-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have a problem with pausing and resuming prints. After I pause and resume, head parks twice, and starts with incorrect z.

### Expected behavior

I expect print resumes at the same Z level

### Steps to Reproduce

1. Printer prints, it is at Z 1.6 (example)
2. I click pause - head parks (240, 230, 20), click Continue.
3. Then I click resume. Head is homed, moved to (0,0) and to correct Z - 1.6
4. Then head is parked again to (240, 230), but is not raised - Z axis stays at 1.6, but the printer shows 20
5. After head is parked and I click Continue, it is again homed to (0,0) and then starts printing - but it thinks it's Z axis is at 20 even though it is at 1.6. It starts lowering and touches bed

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2-422-BLTUBL-MPC, based on bugfix-2.1.x (Dec 22 2022 02:12:07)

### Printer model

Ender 3V2

### Electronics

Board 4.2.2

### Add-ons

Filament sensro, Cr Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

MadTerminator, 2023-02-22:
Same problem with non-ubl version

mriscoc, 2023-02-23:
A new version is cooking right now and that problem can't be replicated, please wait to do the test with the last version.

MadTerminator, 2023-02-23:
By when will it come out?

mriscoc, 2023-02-24:
> By when will it come out?

Maybe the next week.

BradGillies, 2023-02-25:
2.1.2 firmware and same issue.  Cr touch probe does not activate when homing after pause or print completion.  Causes bed crash.

mriscoc, 2023-03-13:
Issue can't be  replicated in latest firmware version.

closed 2023-03-13 by mriscoc

BradGillies, 2023-03-16:
Miguel

 

I have been working on this issue trying to find a cause but I cannot seem to mitigate it.

 

I did not have this issue on the stock Creality firmware.

 

The issue seems to occur when a print is completed or stopped and the machine is left in that state.  If I simply start another print, when the printer executes the G28 to home the print head it will zero X and Y and then proceed to Home Z but will not deploy the Z Probe (CR Touch)

 

If I were to execute a home command from the screen manually the Homing will work as expected.

 

I am pretty new to all of this and still learning the ins and outs of setup so this firmware is a big jump from the stock firmware.  Could there be a setup issue in the firmware that is causing the probe to not deploy when it encounters the G28 command?

 

I think I have it setup correctly but this is the only instance that will have a failure.  Today I forgot to manually home the printer before starting a print and ended up destroying my build surface so that’s not ideal.

 

Any assistance would be appreciated to avoid doing more damage.

 

Thanks

 

Brad Gillies

 

Closed #743 <https://github.com/mriscoc/Ender3V2S1/issues/743>  as completed.

mriscoc, 2023-03-16:
> I have been working on this issue trying to find a cause but I cannot seem to mitigate it.
The issue seems to occur when a print is completed or stopped and the machine is left in that state.  If I simply start another print, when the printer executes the G28 to home the print head it will zero X and Y and then proceed to Home Z but will not deploy the Z Probe (CR Touch)

Not reproducible, the steps that I follow:
1- turn on printer
2- select a file
3- print file
4- wait to the end, and press "Continue"
5- go to media menu and select other file
6- print the new file
Everything works as expected.

Because your issue is not relating to the OP (Pause/Resume) please open other issue.