mriscoc/Ender3V2S1 #749: What this for hyroglyphs?

gorillale, 2023-02-20:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

What this for hyroglyphs

### Steps to Reproduce

Print object

### Version of Professional Firmware

2.2.1

### Printer model

ender 3v2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[image]

gorillale, 2023-02-20:
I used firmware 2022 10 02 and see this after update Repetier Host Pro 1.4.8
I will try the 20221222 now

mriscoc, 2023-02-24:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/359

mriscoc, 2023-02-24:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/328

mriscoc, 2023-03-13:
Closed as duplicate

closed 2023-03-13 by mriscoc