mriscoc/Ender3V2S1 #754: [Improvement] Disable heating while pause

tomaszczura, 2023-02-24:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Not a bug, but an improvement, or restoring behavior I saw on original firmware. 
If I pause the printer, I'd like to have possibility to lower temperature - e.g. to pause it overnight, set bed and nozzle temp to 0 and restore in the morning. Currently it is not possible - on pause, I get a screen with purge/continue buttons. It was possible on original firmware.

### Bug Timeline

old issue

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2-422-BLTUBL-MPC, based on bugfix-2.1.x (Dec 22 2022 02:12:07)

### Printer model

Ender 3V2

### Electronics

Board 4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2023-03-13:
When paused, the heaters are automatically disabled after some time and turned on at resume.

tomaszczura, 2023-03-13:
Does that apply both to bed and nozzle?

auswin17, 2023-03-16:
I agree with @tomaszczura about this feature. When I want to pause overnight I can press pause and make it cooldown before I turn it off. I tried waiting in pause (it shows "Advance Pause" which I understand that it is for filament change) for a long time and it does not seem to cooldown. Do you know how long until the heating system disable automatically?

Still, it would be nice if you can add a cooldown feature while pause, so I can shut it down safely and continue printing in the morning.

Thank you in advance.

closed 2023-05-27 by github-actions[bot]

mriscoc, 2023-07-27:
Latest experimental implements the G-code [M86](https://marlinfw.org/docs/gcode/M086.html) and [M87](https://marlinfw.org/docs/gcode/M087.html) to control the hotend idle timeout