mriscoc/Ender3V2S1 #756: Input Shaping X freq in Y

classicrocker883, 2023-02-24:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

https://github.com/mriscoc/Ender3V2S1/commit/4b086790e9597c7d03e6ffc7c215271aa003f793

### Version of Professional Firmware

Newest

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

classicrocker883, 2023-02-24:
in dwin.cpp file line 3188, fix should be like this https://github.com/mriscoc/Ender3V2S1/commit/4b086790e9597c7d03e6ffc7c215271aa003f793

mriscoc, 2023-02-24:
Fixed in current experimental sources, thanks.
Related to: https://www.facebook.com/groups/ender3s1printer/posts/6526024700759438

mriscoc, 2023-03-13:
Fixed in latest version.

closed 2023-03-13 by mriscoc