mriscoc/Ender3V2S1 #757: Z Homing failing (nozzle touches bed around 2mm)

gsuresh2u, 2023-02-24:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hi, i compiled firmware, configured with your configurator for SKR MINI E3 V3 board for Ender2 V2 
i upgraded to Ender Extender, and added Auto shut down relay, 
there is no opion in precompiled firmware to enable or disable auto shutdown
so, i need to compile my firmware
today i flashed with new firmware which is downloaded today and compiled
when i press Auto Home or Z home, CR Touch touching bed, then not stopping
nozzle is crashing into bed around 2 mm
attached my config files, please check

### Expected behavior

Need to home Z correcttly

### Actual behavior

when Z Homing, nozzle is crashing into bed around 2 mm

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3V2

### Electronics

 SKR MINI E3 V3

### Add-ons

Auto Shudown

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Additional information & file uploads

[config.zip](https://github.com/mriscoc/Ender3V2S1/files/10824755/config.zip)

BradGillies, 2023-02-25:
Similar issue.  After print finishes or halted and Auto home is selected, the z axis will over travel and not stop.  Power must be removed from printer.  The probe does not activate when autohome is selected in the above conditions.  Also the Z stop does not stop the over travel.  I put the Z stop switch back to protect from the over travel to no avail.  This is readily repeatable.

gsuresh2u, 2023-02-25:
But my problem is little different
Z stops after nozzle touching bed around 2mm
Before upgrading firmware, my Z offset is - 0.82
Now it's positive value, 2.6mm
So, there must be firmware problem
I have no backup for my old firmware to update again

gsuresh2u, 2023-02-25:
@mriscoc Please check my problem

mriscoc, 2023-02-25:
Could be related to CR Touch wiring, disable the HS mode.

gsuresh2u, 2023-02-26:
@mriscoc Hi, finally i found the problem, its not CR touch wiring
when i use source code directly from https://github.com/mriscoc/Ender3V2S1, and configured my own settrings
then Nozzle touching bed problem exists
But when i use source code from https://github.com/mriscoc/Ender3V2S1/releases/tag/20221222, the problem is gone
Z is homing correctly, so there must be some glitch in source code https://github.com/mriscoc/Ender3V2S1
i hope you can correct it next release
i am 100% sure there is some glitch, i checked 3 times with flashing firmware 3 times
hope anyone else have same problem, pleae use source from https://github.com/mriscoc/Ender3V2S1/releases/tag/20221222

closed 2023-02-26 by gsuresh2u