mriscoc/Ender3V2S1 #760: Automatic bed PID doesn't terminate

jetcat-bob, 2023-02-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I changed the bed max power variable from 255 to 127 to slow the heating of the bed (500 watt heater). If I do a automatic PID calibrate from room temp to 90c, the calibrate PID looks like it times out but the loop continues to heat beyond the set point (graph still active) and I need to power off to terminate. If I start at a higher temp it works fine.

### Bug Timeline

New

### Expected behavior

Calibrate PID

### Actual behavior

Times out but stuck on loop

### Version of Professional Firmware

12/22

### Printer model

3V2

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-02-25:
For custom compile, try with the current Marlin bugfix branch.

badaszszsz, 2023-03-06:
Could I try to help with the fix somehow? I have Ender 3 v2 (Neo, HW 4.2.2) and could try to bugfix in this trunk, if you want @mriscoc :-) best regards!

jetcat-bob, 2023-03-07:
I probably have a unusual setup that might not be worth investigating this error. I have a 500 watt heater on a 6mm thick build plate that I do not want to heat too quickly. Probably not too many others have this configuration for a Ender3. Lowered the bed max power by 50% so it takes awhile to come up to temperature. My guess the PID loop has a timeout and my heat time exceeds it. Problem is,  it doesn't have a good recovery from this timeout error.

mriscoc, 2023-03-07:
https://github.com/MarlinFirmware/Marlin/issues/12265

closed 2023-05-18 by github-actions[bot]