mriscoc/Ender3V2S1 #762: Extruder speed wrong after esteps correction

mohoo, 2023-02-26:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When setting the esteps and then extruding manually, the extruder seems to use the wrong speed.
Details see my video:

https://www.youtube.com/watch?v=SQ6vexO55Z0

### Bug Timeline

I do not know, I used the firmware only for a few days.

### Expected behavior

The extruder should extrude the filament at the usual speed.

### Actual behavior

The extruder extrudes at what feels like 1000% speed. See my video.

### Steps to Reproduce

1. Change esteps
2. extrude filament

### Version of Professional Firmware

Ender3S1-F4-20221222 (2.13)

### Printer model

Creality Ender 3 S1 F4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2023-02-26:
For manual movements the firmware uses:

```cpp
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
```

So the extruder moves at 2mm/s you can enable axis live movement to have better control of the speed.

mohoo, 2023-02-26:
In general, the speed is correct. I have not adjusted anything else. 
But as soon as I adjust the steps/mm, as in the video and then extrude manually, the extruder shoots the filament through at high speed. If I extrude again afterwards, the speed is correct again.

mohoo, 2023-02-26:
@mriscoc  I have uploaded a second video where you can clearly see that two different speeds are used. I can only reproduce it with certain values. But if I use the values as in the video, then I can reproduce it 100% every time. Can there be a rounding error or something?

The values I use:
X Steps/mm 79.3
Y Steps/mm 79.6
Z Steps/mm 399.7

Move Extruder: 26.3
Important it must happen directly after switching on without homeing, otherwise the error does not occur.

mriscoc, 2023-02-27:
OK thanks, I will review that. It is weird because you don't change extruder steps/mm

mriscoc, 2023-02-27:
The manual feed rate is a constant expresion
https://github.com/mriscoc/Ender3V2S1/blob/8951a70f4b8fdfca3788b99927b3f3b2787b005d/Marlin/src/module/planner.h#L79-L86
So it is not possible to change it at runtime, i.e. the bug you describe could be a problem in the Marlin core. Prior to an axis movement operation, it is advisable to execute an autohome.

mohoo, 2023-02-27:
Ok thanks for checking. 👍

closed 2023-02-27 by mohoo