mriscoc/Ender3V2S1 #770: Momentaneous fail after a print + remove and reload the SD card (Ender S1 F4, UBL 2022 12 22)

UsSuggester, 2023-03-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The problem persist. [https://www.youtube.com/shorts/Eesq9mH4xNU](url)

When I finish a print, I remove the SD card to load another gcode on it, when I insert the SD back to the printer, I want to search for the new gcode by clicking on "Media". When I click on it, the extruder fan stops (while the hotend is still hot so, not good), the knob and the screen freeze for a few seconds (the cursor even appears on an empty slot, down where there are not any [folder).

After a few seconds, the fan start moving normally again, knob and screen get unlocked and appear the system information from "Info Screen", same square with all this; Professional Firmware, Version 2.1.3, MRiscoC, Github link...)

Finish a print, push knob to continue to exit the classic post-printing summary window. (I always do this before remove SD).
Remove the SD card.
Insert SD again.
Go to Media.
Thats it.

Ender 3 S1, F4 (UBL 2022 12 22)

### Version of Professional Firmware

UBL 2022 12 22

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

Jimidean23, 2023-03-01:
I get this reboot as well. I found that it only happens if I move too quickly.  When you insert SD card. Give it a good 10 seconds before you select anything and it won’t reboot or freeze.  That’s been working for me. Sent from my iPhoneJimi Anthony Banda

UsSuggester, 2023-03-01:
> I get this reboot as well. I found that it only happens if I move too quickly.  When you insert SD card. Give it a good 10 seconds before you select anything and it won’t reboot or freeze.  That’s been working for me. 

I will. 
By the way, in my video you can see the message "Media Inserted" and the printer still failed. But I also, and usually, have the "Media Init Fail" when I insert it. Any solution to this?

mriscoc, 2023-03-13:
Format or try with other memory cards.

closed 2023-05-23 by github-actions[bot]