mriscoc/Ender3V2S1 #774: [BUG] Reboot when pressing button

wimg, 2023-03-04:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Using https://github.com/mriscoc/Ender3V2S1/releases/download/20221222/Ender3V2-SKRME3V3-BLTUBL-MPC-20221222.bin on an Ender 3 V2 with the SKR Mini E3 V3.

The systems boots, the display lights up and shows the firmware version and then goes to the menu. I can scroll through the menu by rotating the knob, but as soon as I press the knob, the system hangs, then reboots 5 seconds later.

### Bug Timeline

Just installed the SKR Mini E3

### Expected behavior

Expected the submenu to load

### Actual behavior

System hangs, then reboots

### Steps to Reproduce

1. Installed firmware onto SD card as firmware.bin
2. Powered on system
3. Moved the knob -> moved successfully in menu
4. Pressed button -> hangs and reboots

### Version of Professional Firmware

20221222

### Printer model

Creality Ender 3V2

### Electronics

SKR Mini E3 V3

### Add-ons

BLTouch

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-03-04:
Your screen wiring is not compatible with this firmware, if you have a premade cable check if you did put it inverted.

wimg, 2023-03-05:
I bought the cable that BTT advised me to. If I turn the cable the other way, the screen won't light up.
So I'm really puzzled at this point...

mriscoc, 2023-03-06:
Check here: https://github.com/mriscoc/Ender3V2S1/discussions/131

wimg, 2023-03-10:
Thanks for that, but I bought the cable that is provided for this combination via [https://biqu.equipment/products/mini-e3-connecting-cable-for-ender-3-v2-display-screen?_pos=5&_psq=ENDER3+V2&_ss=e&_v=1.0](https://biqu.equipment/products/mini-e3-connecting-cable-for-ender-3-v2-display-screen?_pos=5&_psq=ENDER3+V2&_ss=e&_v=1.0)

The screen works perfectly, but pressing the button reboots the board. Is there any way to debug what might be going on ?

mriscoc, 2023-03-10:
Check:
[image]

wimg, 2023-03-10:
Wow that must have been the only way I didn't insert them.

Thanks so much !