mriscoc/Ender3V2S1 #775: [BUG] after Mesh leveling Z is incorrect

bronsn, 2023-03-04:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

After leveling the bed with a paper, i use the tramming wizzard to level the bed perfectly.
After that i take a look to Z if the height is ok.
Next step is auto create the mesh and after this step, Z is always incorrect and to high.

if i take my paper after the auto mesh leveling and control the the bed level per hand, the height is always incorrect and the nozzle is to far away from the bed.

### Version of Professional Firmware

 Ender3V2-422-BLTUBL-MPC-20221222.bin 

### Printer model

Creality Ender 3v2

### Electronics

Board 4.4.4

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2023-03-04:
I use this version and it is within the tolerances. your problem could be in your leveling probe.

bronsn, 2023-03-09:
Thx a lot for answering and sry for my late respond but covid smashed me and my fam down ...

classicrocker883, 2023-03-14:
you need to readjust your Z offset. By adding the paper, youre introducing an extra variable, and basically instead of being at level with the bed, the nozzle is level with the paper.

i dont use the paper or gauge method at all, and havent had to relevel or anything after many prints. so what I do is first get the Z offset, this part is important. again, dont use paper. bring the nozzle right down to the center of the bed and stop right when it touches. now when you go to each corner to tram, do basically the same thing except turn the knobs to move the bed up or down. after its about right, go back to the center, is the nozzle touching the bed like before? if not, redo the Z offset, then go ahead and tram/level the bed once again. when Z is 0, you want the nozzle to just be touching the bed surface. i mean **JUST** touching, dont dig into it. u can know by looking or if u feel the fan vibrations on the fan shroud go away since it transfers them into the bed, and dont worry ive done this so many times as long as youre only going up and down, not left or right, you wont scratch anything.

and thats it. as long as when you tell it to go to each corner and center and the JUST starts to touch the bed, youre good. when you start printing, you may still need to adjust your Z offset up or down, but thats talking +/-0.1 or 0.2mm

badaszszsz, 2023-03-17:
@mriscoc do you preserve near to relative or absolute tramming tolerances when used?

I spotted that my CR Probe actuate on +0.07 z offset and Z = 0.0 nozzle need is "touching" bed on z offset -1.51 z 0.0 but until I do retramming and place all nuts just on absolute 0.00/0.01/-0.01 nozzle to probe z offset and z offset would be wrong if set to -1.58.
When I set it in advanced and set it in Z offset outside of it, it looks like it needs absolute tramming to be particularly exceptionally leveled. I used micrometer also and feeler gauges to check. My uncertainty level is about +/-0.005mm

I was doing it other way earlier, setting z offset at the end as most amateur tutorials say.

Wish you all the best and await your or someone else's input about it.

CC: @project-nope

mriscoc, 2023-03-17:
The firmware uses BED_TRAMMING_PROBE_TOLERANCE which is set to 0.05, takes the relative to Z=0 height of each corner and compare them with the average heights to get the relative position of each corner.

Take into account that the advertised mechanical precision of the Ender is 0.1 mm, so, if the head moves and then returns to the previous position an offset of 0.1mm is expected.

badaszszsz, 2023-03-17:
Thanks for telling me and prompt reply!

Before evaluation of those relative, I seen the absolute in probe coordinates the value of it.

I get far better repeatability when moved my bed from absolute probe coordinates 0.12, 0.13, 0.15, 0,14 to near 0.00, and then setting probe offset. 

I need read more code then. And above is probably unrelated. :(

closed 2023-05-28 by github-actions[bot]