mriscoc/Ender3V2S1 #778: [BUG] Turn on the printer is impossible if SD Card is inserted

UsSuggester, 2023-03-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Video: https://www.youtube.com/shorts/2elZnkBZhaE

If the SD card is inserted the printer screen get stucked in the first window, not firmware info square.
In this case notice on the video that the autolevel light is red.

If I turn on the printer without the SD Card inserted, it turn on correctly.
In this case notice that the autolevel light is purple.

### Version of Professional Firmware

Latest December 2022, UBL version for F4

### Printer model

S1 - F4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

badaszszsz, 2023-03-06:
Hey, new one here too. I have heard that sometimes like this happens if you leave the BIN with your firmware there. Or then additional folders are there, and this blocks it.

I had a problem like this with other ("original" software from Creality) 1.1.7 for my Creality 3 v2 Neo. Turns out that connection via COM to computer helped it.

Also, please remember that you actually don't need to "take out" completely the card from the reader. Just push it and leave it.

It work-around your problem (a little), and would be fun if I could help this project in any way in future (first good issue?). I am also open to testing on my printer, as I see programming it now for me is flawless and easy.
I even could rewrite bootloader with in-board interface, if in need.

mriscoc, 2023-03-13:
Try with others memory cards.

earthwormjimwow, 2023-03-16:
I too have this issue, but it seems to only occur once I've crossed some arbitrary quantity of individual files on the SD card.  I can't tell if it is data size related, the quantity of individual files, or perhaps related to the file names or having to parse that much text for all the file names?

Often deleting one file on the SD card, reducing the total quantity by 1 from when it was crashing, resolves the issue.

jimmycav, 2023-04-08:
I've just moved from jyers to mriscoc on my stock ender 3v2 and I also get this issue. I have removed all files from the SD card after upgrading the firmware (mainboard and LCD). If I turn the printer on with the SD card inserted the main menu shows up but cannot move or click anything. The info menu pops up every few seconds.

If I remove the SD card and turn it on it works fine. I can then insert my SD card as normal.

I-wish-2, 2023-04-14:
Greetings,
I'm upgrading all 5 of my various E3 printers to the current firmware release and have noted an SD card related issue although It is not repeatable on any of my printers.  I have 1 SKR Mini3 V3.0 and 4 Ender 4.2.7 boards and this issue has happened on all of them but again I have not been able to get it to repeat consistently.

What I experience is that after printing several STL flies, I remove the SD card and from Cura I save a new one and then re-install the SD card in the printer.

Sometimes the Display screen, Dwin or DACAI, will freeze and not be functional for any dial input.  The firmware then reboots and the printer wakes up fine and I can now select the newly installed STL file and print away.  I did not remove the SD card during this re-boot cycle.

Because this issue is so random for me I suspected that the SD card was in some way the source.  So I did the following:
I re-formatted all of my 8 and 16 gig cards.  I have no more than 4 STL files at a time on any SD card.  I noted when the re-boot occurred and which SD card I was using and who was the manufacture of said SD card trying to pin point the source.  I did not find anything common.

I have not had this issue when flashings a new *.bin file and I have no issue with leaving the SD card with the *.bin file on it in the printer and powering it up.  It always boots up normal.

I reviewed the YouTube video with the E3S1 and can only ask was the SD card's LOCK switched on?  Was the motherboard an F103 or F4 flavor and missed flashed if that is possible.

This re-booting is not an issue for me although it is a curiosity that I would like to pin down.  Till then thanks for reading my comments.
Regards
Doug

closed 2023-06-24 by github-actions[bot]