mriscoc/Ender3V2S1 #783: [BUG] BL touch fails to deploy on 62/64 grid measurement causing head to crash into bed

mikld86, 2023-03-11:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I always run a 64 point measurement for BL touch. 
I normally walk away and let it do its thing. 
Today I stayed and watched.
As I watched, at point 62 or 63 the BL touch failed to deploy, causing the head to crash into the print bed. 

### Expected behavior

BL touch to deploy

### Actual behavior

BL touch failed to deploy

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 S1

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-03-11:
Check the wiki notes about CRTouch you must disable HS mode for meshes larger than 5x5

mikld86, 2023-03-11:
Thanks for pointing me to that solution.

closed 2023-03-11 by mriscoc