mriscoc/Ender3V2S1 #786: [BUG] Auto Build Mesh always seems to be 0.4 higher on the right Row

LunarstarPony, 2023-03-13:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The right row is always been to be 0.4 higher during Auto Build Mesh even tho Tramming says corner leveled. 

### Bug Timeline

It started after updating to the latest Firmware

### Expected behavior

Auto Build mesh being somewhat consistant

### Actual behavior

The very right row on a 5x5 mesh is always 0.4 higher even tho Tramming shows corner leveled

### Steps to Reproduce

Auto Build Mesh

### Version of Professional Firmware

FIRMWARE_NAME:Marlin 2.1.3 MRiscoC Ender3V2-422-BLTUBL-MPC, based on bugfix-2.1.x (Mar 12 2023 17:15:44) SOURCE_CODE_URL:github.com/mriscoc/Ender3V2S1 PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender3V2-422-BLTUBL-MPC EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Ender 3 V2 Neo

### Electronics

4.2.2

### Add-ons

OctoPrint That's all

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

https://youtu.be/HsQuJeaUXMg

LunarstarPony, 2023-03-13:
Printing a test print also shows that it's lifting way too high as well

mriscoc, 2023-03-13:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides
Check if your head is not crashing at right.

gsuresh2u, 2023-03-13:
I have different problem today after updating to today firmware
when i want to build mesh, after homing, X and Y nott moving, its probing at same point, but CR Touch is no touching bed

[config.zip](https://github.com/mriscoc/Ender3V2S1/files/10953722/config.zip)

LunarstarPony, 2023-03-13:
It's not even on the bed actually the nozzle is outside the bed on the right already

> https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides Check if your head is not crashing at right.

mriscoc, 2023-03-13:
> So no it's not crashing, Only the probe is on the bed.

In Neo printers, at very right if you didn't change the X max position as indicated in the calibration guide, the carriage hits a screw slightly.

LunarstarPony, 2023-03-13:
Oh, so it appears that the old value for Max Position I save using M503 before updating doesn't work anymore?(I calibrate it before) It's like 1 off. But it is indeed slightly crashing onto the screw again! Thanks!

closed 2023-03-13 by LunarstarPony

gsuresh2u, 2023-03-13:
when activating UBL, X, Y axis no moving and CR Touch is not touching bed after Homing
it is probing a the same point
But Homing is fine

mriscoc, 2023-03-13:
> when activating UBL, X, Y axis not moving and CR Touch is not touching bed after Homing it is probing at the same point But Homing is fine

Please, open a new issue and fill the issue formulary properly indicating what precompiled firmware and printer are you using.