mriscoc/Ender3V2S1 #787: UBL Failing (Probing at same point)

gsuresh2u, 2023-03-13:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

Him My board is SKR MINI E3 V3 for Ender3-V2 and with Ender extender and with BTT Relay1.2
so, for the reray, its not possible to use precompiled firmware
Today i downloaded this source and compiled for this board
X, Y Z, homing is fine
when i try to Calibrate Mesh,its homing fine, and heating bed to 50
then X, Y axis is not moving, CR Touch is probing at same point, but in screen, its progressing 2/36 , 3/36 , etc...
but after hoimg, CR touch metal pin is not touching Bed, 
i am ataching my config files
[config.zip](https://github.com/mriscoc/Ender3V2S1/files/10954348/config.zip)

### Bug Timeline

Today

### Expected behavior

Need to Calibrate UBL 

### Actual behavior

when Calibrating, UBL, its probing at same point and pin is no ouching bed

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3V2

### Electronics

 SKR MINI E3 V3

### Add-ons

Auto Shudown

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-03-13:
Hi. To discard a problem with your compilation, disable momentary the BTT Relay and flash a probed precompiled firmware.

gsuresh2u, 2023-03-14:
Hi, today i disconnected BTT Relay and flashed with `Ender3V2-SKRME3V3-BLTUBL-MPC-20230312.bin`
and modified bed size to 400x400x450 for my extender kit
but Y axis is homing to front, Z axis is hoiming to Upper, as i replaced Z (dual) and Y axis motors with 2 amp motors
there is no option in screen to revert motors right?
so, there is no chance for me to test with precompiled firmware
or plz compile with Z and Y motors inverting
Thank you

classicrocker883, 2023-03-14:
@gsuresh2u are you using the correct config settings?

in Configuration.h should it be--
>// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true  // BTT SKR MINI E3
#define INVERT_Y_DIR true // BTT SKR MINI E3
#define INVERT_Z_DIR false  // BTT SKR MINI E3

for ender 3 configs, it is false/false/true

and also `#define INVERT_E0_DIR true  // BTT SKR MINI E3` but false for ender 3 configs.

you can check out my [BTT SKR Mini E3 configurations here](https://github.com/classicrocker883/MriscocProUI/tree/Mriscoc-2023/configurations/BTT-SKR_Mini_E3_V3)

i copied them from another source, and I dont have an actual board to test it, but others have used my firmware and I havent heard any negative feedback.

gsuresh2u, 2023-03-14:
Yes, as i rweplaced Y and Z motors, modified like below, then its homing fine
#define INVERT_X_DIR true  // gsuresh2u
#define INVERT_Y_DIR false  // gsuresh2u
#define INVERT_Z_DIR true  // gsuresh2u

gsuresh2u, 2023-03-14:
anyone with the board (SKR MINI E3 V3) having same issue with UBL with precompiled firmware ?

gsuresh2u, 2023-03-15:
Hi, i disabled below
//#define ENDSTOP_NOISE_THRESHOLD 2
//#define ADAPTIVE_STEP_SMOOTHING 
then its working fine

closed 2023-03-15 by gsuresh2u

mriscoc, 2023-03-15:
> Hi, i disabled below
> //#define ENDSTOP_NOISE_THRESHOLD 2
> //#define ADAPTIVE_STEP_SMOOTHING
> then its working fine

But both options are disabled by default if you are using the Configurator. Did you enabled them?

gsuresh2u, 2023-03-15:
Yes