mriscoc/Ender3V2S1 #790: [BUG] Mesh and tramming don't go all the way to the right on a 30x30 bed (ender s1 plus)

glebtv, 2023-03-13:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I've set physical dimensions & axis limits properly.
But tramming and mesh don't go all the way to the right, the last (4th) point for the mesh is here:
[image]
It seems to match the regular ender s1 bed. Can I do anything to change mesh&manual tramming size?

### Version of Professional Firmware

20221222

### Printer model

Ender S1 Plus

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-03-13:
Check mesh insets: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

glebtv, 2023-03-13:
Yes, thank you!
Somehow been skipping over mesh inset settings for 2 days.

closed 2023-03-13 by glebtv